mattjlewis/diozero #1: DigitalInputDevice that doesn't check if a listener is already associated to a pin in the setInputListener()

fpompermaier, 2016-05-31:
Running this with DIO 1.1 

```
try (Button button = new Button(buttonPin, GpioPullUpDown.PULL_UP); LED led = new LED(ledPin)) {
    button.whenPressed(led::on);
    button.whenReleased(led::off);
    SleepUtil.sleepSeconds(10);
}
```

I've received the following exception, caused by DigitalInputDevice that doesn't check if a listener is already associated to a pin in the setInputListener():

```
Exception in thread "main" java.lang.IllegalStateException: The listener is already assigned
	at com.oracle.dio.gpio.impl.GPIOPinImpl.setInputListener(GPIOPinImpl.java:200)
	at com.diozero.internal.provider.jdkdio11.JdkDeviceIoGpioInputDevice.enableListener(JdkDeviceIoGpioInputDevice.java:152)
	at com.diozero.internal.spi.AbstractInputDevice.setListener(AbstractInputDevice.java:48)
	at com.diozero.api.DigitalInputDevice.enableListener(DigitalInputDevice.java:92)
	at com.diozero.api.DigitalInputDevice.whenDeactivated(DigitalInputDevice.java:134)
	at com.diozero.Button.whenReleased(Button.java:62)
```
Best,
Flavio

mattjlewis, 2016-05-31:
Hi Flavio, I will have a look and get back to you ASAP. It looks like the I am calling setInputListener for both whenPressed and whenReleased and DIO 1.1 does not allow that. Most of my testing has been done with the pigpio provider hence I've not encountered this error.

mattjlewis, 2016-05-31:
Ok, I've commited a fix for this issue, thanks for reporting it. I'll be releasing 0.7 shortly.

closed 2016-06-01 by mattjlewis