mattjlewis/diozero #3: Change mode of a pin

fpompermaier, 2016-06-05:
Hi,
in my use case I need to have a list of the pin and the to be able to make them an input or an output at runtime...is that possible? Now when I try to create a new Button or a new Led on an existing pin I get an error like  **Device Native-GPIO-XXX is already in use**.

What is the corret way to achieve this?

Thanks in advance,
Flavio

mattjlewis, 2016-06-05:
Hi, I've been thinking about that scenario for a while given the class APIs are structured heavily towards a single direction.
Currently you have to close the input/output object before you can reopen in a different mode. I've just tested this using the pigpio provider and can confirm that it works. I've added a sample app to github to demonstrate this, please see [InOutTest.java](https://github.com/mattjlewis/diozero/blob/master/diozero-core/src/main/java/com/diozero/sampleapps/InOutTest.java).
However, this solution is a little bit inefficient, I'm considering adding a specific DigitalInOutDevice to improve this - I'd be interested to hear if this is something that you would like and what methods you would consider critical. It wouldn't be as functional as the existing specialised classes such as Button and LED.
Matt

mattjlewis, 2016-06-05:
I meant to state that it is important to invoke ".close()" before reopening that same pin number in a different mode. My example does that implicitly via the try-with-resources block, however, you can achieve the same via calling ".close()" yourself.

fpompermaier, 2016-06-05:
Thanks Matthew,
I'll give you  some feedback asap!

fpompermaier, 2016-06-06:
Is there a way to access the status of the devices (i.e. DeviceStates)? Is there also a way to initialize it wrt the status of exported/enabled pins (NOTE: I'm using udoo quad so you can see if a pin has been exported/enabled if there's a directory for it pin in /sys/class/gpio, e.g. /sys/class/gpio/gpio116/)

mattjlewis, 2016-06-06:
I need to check the portability of this code for checking if the pin is currently exported:
`	/**
	 * Check if this pin is exported by checking the existance of /sys/class/gpio/gpioxxx/
	 * @param pinNumber GPIO pin
	 * @return Returns true if this pin is currently exported
	 */
	public static boolean isExported(int pinNumber) {
		Path path = FileSystems.getDefault().getPath("/sys/class/gpio/gpio" + pinNumber);
		return Files.exists(path) && Files.isDirectory(path);
	}
`
If it works for all providers I will commit it to the DeviceStates class.
Are you encountering issues due to pins having being exported prior to running diozero?
Matt

fpompermaier, 2016-06-07:
Right now I'm using this library to control my udoo https://github.com/BOSSoNe0013/NeoJava but currently it doesn't react to input. However we added a FSWatcher class that detects when files are changed so we can have a real picture of exported gpios. Probably it is not a common use case but I'd like to detect if a some user or program change a pin from input to output or it's state for some reason and, thus, having something like the DeviceStates but that is always in synch with the real situation of the exported gpios, not just what the situation that my program tries to set up (and keep)

mattjlewis, 2016-06-07:
Ok so just a generic utility class that your code can receive events from whenever the GPIO state changes, in particular should some other program do something.
Interesting project BTW, I could implement a file-system based GPIO pure Java solution that would make that library redundant assuming you don't want things like PWM.

fpompermaier, 2016-06-07:
Indeed I see the 2 libraries as complementary: your could be the interface
to the system while the other could be used as a language independent REST
service to interface with the system :)

mattjlewis, 2016-06-19:
I've just commited the sysfs provider that does basic GPIO via the /sys/class/gpio kernel module. The Java WatchService doesn't work for the special /sys file-system and I couldn't see any native Java alternative hence I've been forced to implement my own JNI wrapper around the C poll method. I've compiled the library for raspbian (armhf), no idea whether it will work on other devices.
It would be great if you could give it a go and let me know if it works on your udoo quad. The C code is also on github, if it doesn't work it would be even better if you could compile it and share with me so I can add it back into github. I'll also need to know what value you have for ID in /etc/os-release.

mattjlewis, 2016-12-15:
FYI DigitalInputOutputDevice was added to 0.9-SNAPSHOT

closed 2016-12-15 by mattjlewis

fpompermaier, 2016-12-16:
Great news!