mattjlewis/diozero #6: GPIO input debouncing

fpompermaier, 2016-06-15:
Do you have any strategy to fix this annoying problem?
If yes just tell me where to start and I'll try to issue a PR for it if it's not too complicated

mattjlewis, 2016-06-15:
You could try using the SmoothedInputDevice? I've googled this a few times and not come across a satisfactory answer yet.

fpompermaier, 2016-06-16:
Which is the difference between SmoothedInputDevice and Button?

mattjlewis, 2016-06-16:
Thought you might ask that. Button has the convenience methods isPressed / whenPressed which simply wrap isActive / whenActive. I need to create a class inheritance diagram to illustrate.
One option is for Button to extend SmoothedInputDevice thereby getting debounce support (I think).
Could you try using SmoothedInputDevice and let me know if it provides you the right level of input debouncing. Like I say the only difference the renaming of active / inactive to pressed / released for convenience and representation of button vocabulary.

fpompermaier, 2016-06-22:
I tried to use SmoothedInputDevice  but I didn't find a valuable solution..it seems that events sometimes are not triggered while some other time they are fired 2 or 3 times :(
This happens also with Buttons. Is there a way to get exactly one event on raining/falling edge of the input?

mattjlewis, 2016-06-24:
Have a look at the SmoothedInputTest class in the test package (https://github.com/mattjlewis/diozero/blob/master/diozero-core/src/test/java/com/diozero/SmoothedInputTest.java) - I've just rerun this with some minor changes to the parameters and can see events being smoothed correctly.
What library are you using as the diozero system provider?

fpompermaier, 2016-06-24:
I'm using jdkdio11

mattjlewis, 2016-06-24:
I don't have an issue with Buttons on the Raspberry Pi, using any of the providers. It could be an issue with JDK Device I/O - do you have code where this does work?
Check this page for my wiring and code that I use to validate on the Pi: http://rtd.diozero.com/en/latest/InputDevices/#input-devices

fpompermaier, 2016-06-24:
You're right, using  button.whenPressed() worked like a charm :+1:

closed 2016-06-24 by fpompermaier