mattjlewis/diozero #7: LCD set Text

carduque, 2016-06-24:
Hi,
I created an I2C LCD like that: `new I2CLcd(16, 2)`

When I set text 
`lcd.setText(0, "Hello World); `
it always throw this error:

```
Exception in thread "main" java.lang.StringIndexOutOfBoundsException: String index out of range: 16
        at java.lang.String.substring(String.java:1950)
        at com.diozero.I2CLcd.setText(I2CLcd.java:348)
```

What am I doing wrong?
Thanks.

mattjlewis, 2016-06-24:
Apologies, no idea how that error get there! This was fixed in 0.8-SNAPSHOT but I've only just pushed this snapshot version to the repositories. I have also updated the separate binaries on Google drive. FYI I will be pushing the final 0.8 release within a few days.

closed 2016-06-24 by mattjlewis

carduque, 2016-06-24:
It works. Thanks!