mattjlewis/diozero #9: Don't restrict to NCP PCF8574 chip

Swinkid, 2016-10-28:
Hi,

Any chance we can get other expander boards apart from the PCF chip?

I'm using a MCP23008 chip I'd like to use with your lib

mattjlewis, 2016-10-28:
Sure, I'll look into it this weekend.

mattjlewis, 2016-10-28:
Just pushed an untested class to github, not updated any repository builds yet. Let me know how you get on with it - looking at the datasheet it was (unsurprisingly) very similar to the MCP23017. If it works I'll work on merging the 2 classes as they are so similar.

Swinkid, 2016-10-28:
Cheerrs buddy. I'll try and import it sometime tomorrow. Thanks for the quick implementation!

Swinkid, 2016-12-13:
Hey,

Completely forgot about this. I apologies. I was just wondering, Is it possible to use your library to write to a LCD over i2c with the MCP23008 chip?

Cheers

mattjlewis, 2016-12-14:
I've only tested LCDs fitted with a PCF8574T I2C backpack (see the second image here: http://www.raspberrypi-spy.co.uk/2015/05/using-an-i2c-enabled-lcd-screen-with-the-raspberry-pi/). The code probably won't currently work with the MCP23008 as it uses a slightly different I2C communication protocol. Should be fairly straightforward to refactor the code to work with different IO expanders if you fancy a challenge. I only have an MCP23017, I could try unsolder one of my LCDs and try but won't have time to do this until after Christmas.

mattjlewis, 2016-12-18:
I've just committed _untested_ changes that may allow use of I2C LCD displays connected via an MCP23008. Use the `I2CLcd(GpioExpander gpioExpander, int columns, int rows)` contructor, i.e. `new I2CLcd(new MCP23008(), columns, rows);`, New 0.9-SNAPSHOT builds have been pushed to sonartype Maven snapshot repositories.
I need to regression test my I2CLcds to make sure I've not broken anything. It will take a bit of effort for me test with an MCP23008 not least because I only have an MCP23017 (but they are very similar). Let me know how you get on with it, fingers crossed.

Swinkid, 2016-12-18:
Sweet, I'll give it a poke. Will the snapshot be on Nexus?

Swinkid, 2016-12-18:
May be being stupid here, but getting this exception:

`Exception in thread "main" java.lang.IllegalStateException: Error: no device provider factory service found, please configure META-INF/services/com.diozero.internal.spi.NativeDeviceFactoryInterface or set -Dcom.diozero.devicefactory
	at com.diozero.internal.DeviceFactoryHelper.init(DeviceFactoryHelper.java:80)
	at com.diozero.internal.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:88)
	at com.diozero.api.I2CDevice.<init>(I2CDevice.java:66)
	at com.diozero.MCP23008.<init>(MCP23008.java:130)
	at com.diozero.MCP23008.<init>(MCP23008.java:120)
	at com.alex.noble.i2c.MainTest.main(MainTest.java:16)`

`    public static void main(String[] args) throws Exception {

        MCP23008 mcp = new MCP23008();

        I2CLcd lcd = new I2CLcd(mcp, 16, 2);

        lcd.addText("Testing");
    }`

Apologies if its something painfully obvious, been programming all day! :)

mattjlewis, 2016-12-18:
Ok, I can confirm that the LCD code still works on my setup, command:
`sudo java -cp tinylog-1.1.jar:diozero-core-0.9-SNAPSHOT.jar:diozero-provider-sysfs-0.9-SNAPSHOT.jar com.diozero.sampleapps.I2CLcdSampleApp16x2`
Looks like you've not included a provider JAR file in your classpath. diozero is designed to work with different implementations, including Pi4j, WiringPi, pigpio plus some limited but pure Java ones that I've developed. I guess I should include my one within the main JAR file as a default to simplify the classpath. Just need to add SPI support and do further testing,

Swinkid, 2016-12-18:
I'll re add Pi4J and let you know. Mistake on my behalf - totally didn't read the instructions properly :)

mattjlewis, 2016-12-18:
Good news is that I had a very old [PiFace ](www.piface.org.uk/) kicking around that I'd never used before. This has an LCD screen controlled by an MCP23S17. The bad news is that this is different to your configuraiton plus I don't have code for the SPI equivalent of the MCP23017. Job for another day... Good luck.

mattjlewis, 2017-02-06:
did you get anywhere with this? I've just committed changes that should make this much easier. There is a new LcdConnection interface that gives you programmatic control over how you communicate with the LCD. I have tested this successfully with the MCP23S17 (SPI) included in the PiFace Control and Display board. Should be trivial to do this with a MCP23008 (I2C).

mattjlewis, 2017-02-27:
Marking this issue as closed; I have successfully tested LCDs connected via MCP23S17. I should now be trivial to connect via other means, including GPIO.

closed 2017-02-27 by mattjlewis