mattjlewis/diozero #12: I2C bit banging

tnogueira, 2016-12-14:
Hi,

First of all thanks for the great library. I use the pigpio Python library, and i find the I2C bit banging functions very useful. They don't seem to be included in the current release (?) Are there any plans to bring them into diozero?

Thanks
Tiago

mattjlewis, 2016-12-14:
Looks quite straightforward, I'll seee what I can do.

mattjlewis, 2016-12-14:
I've made the changes locally, not had time to test it though. Will do so ASAP and commit once done.

tnogueira, 2016-12-15:
Great, thanks for the quick reply!

mattjlewis, 2016-12-15:
To be clear I've added this to the pigpio Java wrapper only at this stage. I don't think it will be possible to add this in a device agnostic fashion as it would be extremely difficult to add to all other providers assuming that most if not all of them don't support this.

tnogueira, 2016-12-15:
I think that's reasonable, and works for me.

mattjlewis, 2016-12-15:
Finally had a chance to simply check the code compiles which it does. Have committed changes to github, plus updated the compiled library. Note this is currently untested, please let me know of any issues. I will test with an I2C LCD display over the weekend.

tnogueira, 2016-12-16:
Hi,

I'm probably missing something here, but i don't seem to find the implementation. The 'diozero-provider-pigpio' has not been updated with your last commit.

Thanks again
Tiago

tnogueira, 2016-12-16:
Sorry, just realised you have updated the 'pigpioj' project. I'll give it a try.
Thanks!

mattjlewis, 2016-12-17:
Yes, I should have said. I've incorporated the equivalent but slightly more useable wrapper methods into the pigpio provider. Will commit the changes today. Let me know of any issues.

tnogueira, 2016-12-17:
Hi,

I've tried it, and i always get the error: PI_BAD_I2C_RLEN. I can see that the write operation works, but then something goes wrong with the reading. The same transaction over the Python wrapper works just fine. 

The set of operations, a repeated start transaction, is as follows: [0x04, 0x62, 0x02, 0x07, 0x01, 0x01, 0x02, 0x06, 0x32, 0x03, 0x00].

Still trying to figure it out...

tnogueira, 2016-12-17:
Also there seems to be a small typo/bug on the pigpio C library:
https://github.com/joan2937/pigpio/blob/master/pigpio.c#L10385

It should probably read PI_BAD_I2C_WLEN and not PI_BAD_I2C_RLEN

mattjlewis, 2016-12-17:
I wonder if I need to set the ByteOrder...

tnogueira, 2016-12-19:
Thanks, i'll give it a try this evening. Btw, for my implementation i'm using the pigpioj code directly, and not the diozero wrapper.

About the wrong error code. It looks to me like a problem with the actual pigpio C implementation, and not your wrapper. Would you agree?

tnogueira, 2016-12-19:
So,  i got it work. The problem was not with the endianness but rather with a small "mismatch" in the expected buffer sizes. Looking at the C implementation in pigpio.c (https://github.com/joan2937/pigpio/blob/master/pigpio.c#L10351) the outBuf must be at least 1 byte larger than the number of bytes you want to read. I think this line should rather read: "if ((bytes + outPos) <= outLen)", otherwise the behaviour is a bit unexpected.

mattjlewis, 2016-12-20:
Good work, let me know if I need to make any changes otherwise I'll close this issue.

tnogueira, 2016-12-20:
On pigpioj i wouldn't change anything. On the diozero-provider-pigpio you can remove the line that sets the endianness. Otherwise, keep up the good work, and thanks again!

mattjlewis, 2017-01-02:
Done, thanks for testing this out. I'm always happy to receive feedback and suggestions.

closed 2017-01-02 by mattjlewis