mattjlewis/diozero #14: Support for Java 1.7

frizman21, 2017-02-19:
Hey - I noticed that you don't support Java 1.7 - I have a project that for, a specific reason of which I'll blame a nameless Fortune 100 company for not also moving to 1.8, I'm still running 1.7. I pulled your latest release and gave it a quick try - seems that you're using some 1.8 language features.

mattjlewis, 2017-02-20:
Hi, yes I deliberately used Java 8 features where available and make development easier. I've never tried compiling or running with Java 7, let me know how you get on - I'm happy to make some small changes. 
I'd be interested to know your use case and device.

frizman21, 2017-02-21:
CHIP + Google App Engine, which is presently stuck on Java 1.7

frizman21, 2017-02-27:
So far so good. Be free to close. I'll resurrect if I find something I can't otherwise overcome.

closed 2017-03-04 by mattjlewis