mattjlewis/diozero #15: CHIP BoardInfoProvider could not be instantiated

frizman21, 2017-02-21:
I've got the LED output working great. I attempted a api.DigitalInputDevice (or a Button) and got the following error on the constructor call. It seems that it's trying to find some Raspberry Pi board information, while I'm running a CHIP. Please advise...

`Exception in thread "main" java.util.ServiceConfigurationError: com.diozero.util.BoardInfoProvider: Provider com.diozero.internal.board.raspberrypi.RaspberryPiBoardInfoProvider could not be instantiated
	at java.util.ServiceLoader.fail(ServiceLoader.java:232)
	at java.util.ServiceLoader.access$100(ServiceLoader.java:185)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:384)
	at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
	at com.diozero.util.SystemInfo.lookupBoardInfo(SystemInfo.java:78)
	at com.diozero.util.SystemInfo.initialise(SystemInfo.java:71)
	at com.diozero.util.SystemInfo.getBoardInfo(SystemInfo.java:112)
	at com.diozero.util.SystemInfo.getLibraryPath(SystemInfo.java:118)
	at com.diozero.util.LibraryLoader.loadLibrary(LibraryLoader.java:44)
	at com.diozero.util.PollNative.<clinit>(PollNative.java:32)
	at com.diozero.internal.provider.sysfs.SysFsDigitalInputDevice.<init>(SysFsDigitalInputDevice.java:67)
	at com.diozero.internal.provider.sysfs.SysFsDeviceFactory.createDigitalInputPin(SysFsDeviceFactory.java:97)
	at com.diozero.internal.spi.BaseNativeDeviceFactory.provisionDigitalInputPin(BaseNativeDeviceFactory.java:108)
	at com.diozero.api.DigitalInputDevice.<init>(DigitalInputDevice.java:90)
	at com.diozero.api.DigitalInputDevice.<init>(DigitalInputDevice.java:70)
	at com.diozero.Button.<init>(Button.java:66)
	at [my code]
Caused by: java.lang.ExceptionInInitializerError
	at com.diozero.internal.board.raspberrypi.RaspberryPiBoardInfoProvider.<clinit>(RaspberryPiBoardInfoProvider.java:150)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:380)
	... 19 more
Caused by: java.lang.UnsupportedOperationException
	at java.util.AbstractList.add(AbstractList.java:148)
	at java.util.AbstractList.add(AbstractList.java:108)
	at com.diozero.internal.board.raspberrypi.RaspberryPiBoardInfoProvider$PiBRev1BoardInfo.<clinit>(RaspberryPiBoardInfoProvider.java:281)
	... 26 more`

frizman21, 2017-02-21:
upgrade to 0.9-SNAPSHOT

closed 2017-02-21 by frizman21

mattjlewis, 2017-02-21:
Yes, sorry about that. I really need to push a new build. Will do v0.9 this week.