mattjlewis/diozero #16: HCSR04 on BeagleBone doesn't work

asuar078, 2017-03-06:
Hello, I was trying to use the HCSR04 sensor on the beaglebone black (Debian wheezy kernel 3.8.13-bone80) but I keep getting the same error. 

> 20:40:01.741 [main] ERROR com.diozero.HCSR04.getDistanceCm - Timeout exceeded waiting for echo to go high

A quick look on my scope showed the issue to be that the trigger pin isn't creating a 10us pulse, instead, the pulse seems to be much larger ranging from 50-100us long. 

This is strange since I tried the same sensor on the rpi3 and had no issues. 

Any clues as to where the problem is coming from?

asuar078, 2017-03-06:
Code running 

`try (HCSR04 hcsr04 = new HCSR04(67, 66)) {
            while(true){
                double distance_cm = hcsr04.getDistanceCm();
                System.out.println("Distance : + " + distance_cm + " cm");
            }
        } finally {
            DeviceFactoryHelper.getNativeDeviceFactory().shutdown();
        }`

mattjlewis, 2017-03-11:
Sorry for the delayd response. I'll look into this today / tomorrow. What Java and diozero version are you using?
Matt

AsuarLinux, 2017-03-11:
On the beaglebone I am using java version 1.8.0_77 and the latest diozero that is available on the maven repository which is 0.9.

asuar078, 2017-03-11:
On the beaglebone I am using java version 1.8.0_77 and the latest diozero that is available on the maven repository which is 0.9.

mattjlewis, 2017-03-12:
Ok, I think I know what the problem is. The library defaults to the sysfs provider; I guess it is just too slow. I will create a new wrapper project around this C library (https://github.com/VegetableAvenger/BBBIOlib/tree/master/BBBio_lib) that uses direct memory writes rather than sysfs.
Will let you know once done, won't take long.
Matt

asuar078, 2017-03-12:
Sounds great thanks!

mattjlewis, 2017-03-12:
FYI I'm running:Linux beaglebone 4.4.36-ti-r71 #1 SMP Sun Dec 4 02:56:26 UTC 2016 armv7l GNU/Linux
Just committed my changes and pushed a new snapshot build. Note I've not tested it yet, let me know how you get on.
Matt

mattjlewis, 2017-03-12:
And make sure you include diozero-provider-bbbiolib-0.10-SNAPSHOT.jar on the classpath. 

Regards,
Matt

asuar078, 2017-03-15:
Appears to be working, Thanks!

mattjlewis, 2017-03-16:
Excellent, thanks for testing and letting me know.
Matt

closed 2017-03-16 by mattjlewis