mattjlewis/diozero #24: Using GPIO pin 10 (SPI) with WS281x java wrapper

mat0pad, 2017-10-09:
First of really awesome lib you've made here!

I'm trying to use SPI instead of PCM and PWM (need analog and digital audio). But when I change the pin number to 10 (after SPI is enabled) I get the following error:

`java.lang.NullPointerException
	at com.diozero.ws281xj.WS281x.<init>(WS281x.java:104)
	at com.diozero.ws281xj.WS281x.<init>(WS281x.java:88)
	at com.ibm.bluesmarties.Example.Ws281xExample.main(Ws281xExample.java:18)
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x632cca54, pid=937, tid=1663870064
#
# JRE version: Java(TM) SE Runtime Environment (8.0_65-b17) (build 1.8.0_65-b17)
# Java VM: Java HotSpot(TM) Client VM (25.65-b01 mixed mode linux-arm )
# Problematic frame:
# C  [libws281xj7159970522658493473.so+0x1a54]  ws2811_wait+0xc
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
JNI_OnLoad()
W281xj_initialise()

bash: line 1:   933 Aborted                 sudo java -cp classes:lib/'*' Example.Ws281xExample`

From the stacktrace is seems to go wrong here:
`public WS281x(int var1, int var2, int var3, int var4, int var5) {
        init();
        this.numPixels = var5;
        this.ch0LedBuffer = WS281xNative.initialise(var1, var2, var3, var4, var5);
        System.out.println("order=" + this.ch0LedBuffer.order()); // ch0LedBuffer is NULL
        this.ch0LedBuffer.order(ByteOrder.LITTLE_ENDIAN);
    }`

mattjlewis, 2017-10-09:
Hi sorry about this, I've just been through the code and I've made some adjustments plus upgraded to the latest version of the rpi_ws281x library. I've also improved some of the error handling in the native library.
I've validated that the code runs correctly on my Pi2 without any LEDs connected and have pushed a new snapshot build to Maven central (plus updated the distribution ZIP on Google Drive). When I get a moment I will dig out one of my LED strips and test properly - could be good to get this going again in time for Halloween ;-)
Please let me know how you get on.

mat0pad, 2017-10-09:
This is great and thanks for the quick response!  I will try it out first thing tomorrow and let you know how it goes.

mat0pad, 2017-10-11:
GPIO 10 and SPI works to control neopixels. Thanks for the help!

closed 2017-10-11 by mat0pad