mattjlewis/diozero #25: Is there an option to use channel 1 instead of zero?

yohayg, 2017-10-29:
I'm using GPIO pin 13 which will work when the channel is 1 and not zero.

There is no option to pass the channel number in the native.

mattjlewis, 2017-10-29:
Hi, what aspect of diozero are you referring to here?

yohayg, 2017-10-29:
diozero-ws281x-java. I'm trying to use WS281xTest.java to set the channel same as in https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/strandtest.py

mattjlewis, 2017-10-29:
I've just committed code that hopefully fixes this. Could you test and close the issue if it works please (I need to dig out my neopixels...).
Thanks, Matt

yohayg, 2017-10-29:
Great, thanks I'll check. (I have just finished building my own jni for setting the channel). I'll test it first thing tomorrow!

yohayg, 2017-10-30:
OK. The only thing I had to fix in com_diozero_ws281xj_WS281x.c from

	jobject direct_buffer = (*env)->NewDirectByteBuffer(env, led_string.**channel[0]**.leds, numLeds*sizeof(ws2811_led_t)); 

to 
	jobject direct_buffer = (*env)->NewDirectByteBuffer(env, led_string.**channel[channel]**.leds, numLeds*sizeof(ws2811_led_t));

mattjlewis, 2017-10-30:
Great, thanks for testing. I've committed that change, I'll push a new snapshot build tonight and close this issue when done.

closed 2017-11-03 by mattjlewis