mattjlewis/diozero #26: MFRC522 Example

balferreira, 2017-12-09:
Hi there.
I'm trying to use your Lib, more exactly the MFRC522 interface but I can really understand what are the right parameters to pass as arguments here: <spi-controller> <chip-select> <rst-gpio>

I trying this example and I've this connection http://www.instructables.com/id/Raspberry-Pi-3-Model-B-MIFARE-RC522-RFID-Tag-Readi/ . And works great. 

Can you tell me that are the right parameters to pass to your MFRC522Test?

Kind regards

balferreira, 2017-12-09:
I guess I made it. But now I'm facing another problem. Please check:

```
pi@raspberrypi:~/DioZero $ sudo ./boot.sh 0 0 22
10:12:47.194 [main] DEBUG com.diozero.util.DeviceFactoryHelper.init - Using native device factory class SysFsDeviceFactory
10:12:47.364 [main] DEBUG com.diozero.internal.provider.sysfs.SysFsDeviceFactory.export - export(22, DIGITAL_OUTPUT)
10:12:47.596 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
10:12:47.598 [main] DEBUG com.diozero.internal.provider.sysfs.NativeSpiDevice.<init> - Opening /dev/spidev0.0, frequency 1000000 Hz, mode MODE_0
10:12:47.623 [main] DEBUG com.diozero.devices.MFRC522.init - reset pin was off
10:12:47.721 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - timer interrupt, n: 0x45
10:12:47.722 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - Timed out waiting for interrupt
10:12:48.762 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - timer interrupt, n: 0x45
10:12:48.763 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - Timed out waiting for interrupt
10:12:49.790 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - timer interrupt, n: 0x45
10:12:49.792 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - Timed out waiting for interrupt
10:12:50.794 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - FIFO Level: 2
10:12:50.795 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - ok
A card is present!
10:12:50.802 [main] DEBUG com.diozero.devices.MFRC522.select - uid_complete: false, cascade_level: 1
10:12:50.826 [main] DEBUG com.diozero.devices.MFRC522.select - select_done: false
10:12:50.828 [main] DEBUG com.diozero.devices.MFRC522.select - ANTICOLLISION: current_level_known_bits=0
10:12:50.829 [main] DEBUG com.diozero.devices.MFRC522.select - tx_last_bits: 0
10:12:50.832 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - FIFO Level: 5
10:12:50.833 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - ok
10:12:50.834 [main] DEBUG com.diozero.devices.MFRC522.select - select_done: false
10:12:50.835 [main] DEBUG com.diozero.devices.MFRC522.select - SELECT: current_level_known_bits=32
10:12:50.837 [main] DEBUG com.diozero.devices.MFRC522.select - tx_last_bits: 0
10:12:50.841 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - FIFO Level: 3
10:12:50.843 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - ok
10:12:50.853 [main] DEBUG com.diozero.devices.MFRC522.select - End of while (!uidComplete) loop, uid: UID [uidBytes=108E30A4, sak=8]
Scanned PICC's UID: {108E30A4}
10:12:50.857 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - FIFO Level: 1
10:12:50.859 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - ok
10:12:51.860 [main] DEBUG com.diozero.api.SpiDevice.close - close()
10:12:51.862 [main] DEBUG com.diozero.internal.provider.AbstractDevice.close - close(), key=Native-SPI-0-0
10:12:51.864 [main] DEBUG com.diozero.internal.provider.sysfs.SysFsSpiDevice.closeDevice - closeDevice()
10:12:51.866 [main] DEBUG com.diozero.internal.DeviceStates.closed - closed(Native-SPI-0-0)
10:12:51.871 [DIO-Zero Shutdown Handler] DEBUG com.diozero.util.ShutdownHandlerThread.run - Shutdown handler running
10:12:51.877 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.provider.AbstractDeviceFactory.close - close()
10:12:51.878 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.DeviceStates.closeAll - closeAll()
10:12:51.887 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.provider.AbstractDevice.close - close(), key=Native-GPIO-22
10:12:51.888 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.provider.sysfs.SysFsDigitalOutputDevice.closeDevice - closeDevice()
10:12:51.892 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.DeviceStates.closed - closed(Native-GPIO-22)
10:12:51.894 [DIO-Zero Shutdown Handler] DEBUG com.diozero.util.ShutdownHandlerThread.run - Shutdown handler finished
pi@raspberrypi:~/DioZero $ 
```

After reading one card, the application exist without any reason. Can you tell me why this happening ?

Kind regards.

mattjlewis, 2017-12-10:
What version of diozero is this? I'd recommend the latest 0.11-SNAPSHOT version.
Regards,Matt

balferreira, 2017-12-10:
Yes, I'm using the diozero-core-0.11-SNAPSHOT.
And this is the code that I'm testing: https://github.com/mattjlewis/diozero/blob/master/diozero-sampleapps/src/main/java/com/diozero/sampleapps/mfrc522/MFRC522Test.java

Kesaya, 2018-01-16:
Hey there,
first of all thank you for this great api.
I also got the Sample working and it reads my tag just fine. Then when i copy sample code to my project and run my applicaiton with the same libraries it faisl. I assume I'm doing something wrong when referencing the necessary libraries but I dont know what. I know this is probably a noob question...
this works:
java -cp tinylog-1.2.jar:diozero-core-0.11-SNAPSHOT.jar:diozero-sampleapps-0.11-SNAPSHOT.jar com.diozero.sampleapps.sandpit.MFRC522Test 0 0 25

this doesn't:
java -cp tinylog-1.2.jar:diozero-core-0.11-SNAPSHOT.jar:myapp.jar myapp.MFRC522Test 0 0 22

do i need to reference additional libraries when running my own application? 
The error i get is: java.util.ServiceConfigurationError: com.diozero.internal.provider.NativeDeviceFactoryInterface: Provider com.diozero.internal.provider.bbbiolib.BbbIoLibDeviceFactory could not be instantiated
...
Caused by java.lang.UnsatisfiedLinkError: com.diozero.internal.provider.bbbiolib.BbbIoLibNativ.init()I

Thanks you for any help in advance

Best regards

mattjlewis, 2018-01-17:
Hi Kesaya,

Does your myapp.jar file have a META-INF/services/com.diozero.internal.provider.NativeDeviceFactoryInterface file?

That is the only way that the BbbIoLibDeviceFactory class could be referenced.

Regards,
Matt

Leo-Virgolini, 2018-04-12:
Hello,

I was able to run the _MFRC522_ correctly on the _Raspberry Pi3_ but not on an **Asus Tinker Board**.

Wiring:
 * SDA:  SPI0 CE0 (GPIO8)
 * SCK:  SPI0 SCLK (GPIO11)
 * MOSI: SPI0 MOSI (GPIO10)
 * MISO: SPI0 MISO (GPIO9)
 * IRQ:  Not connected
 * GND:  GND
 * RST:  Any free GPIO (GPIO25)
 * 3v3:  3v3
 
`new MFRC522(0, 0, 25);`

How do I do it?

Thanks!

mattjlewis, 2018-05-14:
This is likely the GPIO number that you are using as the RST pin. See the GPIO pinout table towards the end of [ASUS TinkerBoard website](https://www.asus.com/uk/Single-Board-Computer/Tinker-Board/). diozero uses the GPIO.ASUS column values.

FirasKORKAD, 2018-05-28:
Hello,
I thank you for this library.
I tried to put tinylog and diozero-core-0.11 in pom.xml in order to run a single .JAR but unfortunately it does not work. could you help me solve this problem.
Best regards.

closed 2020-02-11 by mattjlewis