mattjlewis/diozero #27: BBB, error closing device

izhidov, 2017-12-29:
Running on BBB, testing GPIO 67 getting the following after main part executes

java -cp lib/tinylog-1.2.jar:lib/diozero-core-0.11-SNAPSHOT.jar:lib/diozero-sampleapps-0.11-SNAPSHOT.jar:lib/diozero-provider-jdkdio11-0.11-SNAPSHOT.jar:lib/dio-1.1-dev-linux-armv6hf.jar com.diozero.sampleapps.LEDTest 67
03:41:10.993 [main] DEBUG com.diozero.util.DeviceFactoryHelper.init - Using native device factory class SysFsDeviceFactory
03:41:11.309 [main] DEBUG com.diozero.internal.provider.sysfs.SysFsDeviceFactory.export - export(67, DIGITAL_OUTPUT)
03:41:11.370 [main] INFO com.diozero.sampleapps.LEDTest.test - On
03:41:11.589 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
03:41:12.617 [main] INFO com.diozero.sampleapps.LEDTest.test - Off
03:41:13.626 [main] INFO com.diozero.sampleapps.LEDTest.test - Toggle
03:41:14.634 [main] INFO com.diozero.sampleapps.LEDTest.test - Toggle
03:41:15.641 [main] INFO com.diozero.sampleapps.LEDTest.test - Blink 10 times
03:41:25.657 [main] INFO com.diozero.sampleapps.LEDTest.test - Done
03:41:25.665 [main] DEBUG com.diozero.api.DigitalOutputDevice.close - close()
03:41:25.669 [main] DEBUG com.diozero.internal.provider.AbstractDevice.close - close(), key=Native-GPIO-67
03:41:25.676 [main] DEBUG com.diozero.internal.provider.sysfs.SysFsDigitalOutputDevice.closeDevice - closeDevice()
03:41:25.689 [main] ERROR com.diozero.internal.provider.AbstractDevice.close - Error closing device Native-GPIO-67: com.diozero.util.RuntimeIOException: java.io.IOException: Invalid argument: com.diozero.util.RuntimeIOException: java.io.IOException: Invalid argument
        at com.diozero.internal.provider.sysfs.SysFsDeviceFactory.unexport(SysFsDeviceFactory.java:225)
        at com.diozero.internal.provider.sysfs.SysFsDigitalOutputDevice.closeDevice(SysFsDigitalOutputDevice.java:111)
        at com.diozero.internal.provider.AbstractDevice.close(AbstractDevice.java:66)
        at com.diozero.api.DigitalOutputDevice.close(DigitalOutputDevice.java:102)
        at com.diozero.sampleapps.LEDTest.test(LEDTest.java:90)
        at com.diozero.sampleapps.LEDTest.main(LEDTest.java:68)
Caused by: java.io.IOException: Invalid argument
        at java.io.FileOutputStream.writeBytes(Native Method)
        at java.io.FileOutputStream.write(FileOutputStream.java:326)
        at sun.nio.cs.StreamEncoder.writeBytes(StreamEncoder.java:221)
        at sun.nio.cs.StreamEncoder.implClose(StreamEncoder.java:316)
        at sun.nio.cs.StreamEncoder.close(StreamEncoder.java:149)
        at java.io.OutputStreamWriter.close(OutputStreamWriter.java:233)
        at com.diozero.internal.provider.sysfs.SysFsDeviceFactory.unexport(SysFsDeviceFactory.java:224)
        at com.diozero.internal.provider.sysfs.SysFsDigitalOutputDevice.closeDevice(SysFsDigitalOutputDevice.java:111)
        at com.diozero.internal.provider.AbstractDevice.close(AbstractDevice.java:66)
        at com.diozero.api.DigitalOutputDevice.close(DigitalOutputDevice.java:102)
        at com.diozero.sampleapps.LEDTest.test(LEDTest.java:90)
        at com.diozero.sampleapps.LEDTest.main(LEDTest.java:68)
03:41:25.712 [main] DEBUG com.diozero.internal.DeviceStates.closed - closed(Native-GPIO-67)
03:41:25.812 [main] DEBUG com.diozero.internal.provider.AbstractDeviceFactory.close - close()
03:41:25.821 [main] DEBUG com.diozero.internal.DeviceStates.closeAll - closeAll()
03:41:25.879 [DIO-Zero Shutdown Handler] DEBUG com.diozero.util.ShutdownHandlerThread.run - Shutdown handler running
03:41:25.888 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.provider.AbstractDeviceFactory.close - close()
03:41:25.893 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.DeviceStates.closeAll - closeAll()
03:41:25.895 [DIO-Zero Shutdown Handler] DEBUG com.diozero.util.ShutdownHandlerThread.run - Shutdown handler finished

mattjlewis, 2018-01-17:
Hi, sorry for the delay replying. Could you try run this without the diozero-provider-jdkdio11-0.11-SNAPSHOT.jar and dio-1.1-dev-linux-armv6hf.jar JAR files on the classpath.
I've not tested using JDK device I/O on the BBB - only the sysfs implementation included in diozero-core and the BBIOlib implementation.

Regards,
Matt

drogga, 2019-03-24:
I get this same error without the jar on the classpath 

See command and output below 

 java -classpath tinylog-1.2.jar:diozero-core-0.11-SNAPSHOT.jar:diozero-sampleapps-0.11-SNAPSHOT.jar com.diozero.sampleapps.ButtonTest 66

14:30:05.748 [main] DEBUG com.diozero.util.DeviceFactoryHelper.init - Using native device factory class SysFsDeviceFactory
14:30:06.005 [main] DEBUG com.diozero.internal.provider.sysfs.SysFsDeviceFactory.export - export(66, DIGITAL_INPUT)
14:30:06.284 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
14:30:06.373 [diozero-EpollNative-processEvents-9635026] INFO com.diozero.sampleapps.ButtonTest.lambda$0 - Pressed
14:30:06.387 [diozero-EpollNative-processEvents-9635026] INFO com.diozero.sampleapps.ButtonTest.lambda$2 - valueChanged(DigitalInputEvent [gpio=66, epochTime=1549117806370, nanoTime=20211893121741, value=false])
14:30:06.380 [main] DEBUG com.diozero.sampleapps.ButtonTest.test - Waiting for 10s - *** Press the button connected to input pin 66 ***
14:30:08.358 [diozero-EpollNative-processEvents-9635026] INFO com.diozero.sampleapps.ButtonTest.lambda$1 - Released
14:30:08.360 [diozero-EpollNative-processEvents-9635026] INFO com.diozero.sampleapps.ButtonTest.lambda$2 - valueChanged(DigitalInputEvent [gpio=66, epochTime=1549117808357, nanoTime=20213880623616, value=true])
com.diozero.sampleapps.ButtonTest.lambda$0 - Pressed
14:30:16.012 [diozero-EpollNative-processEvents-9635026] INFO com.diozero.sampleapps.ButtonTest.lambda$2 - valueChanged(DigitalInputEvent [gpio=66, epochTime=1549117816009, nanoTime=20221532552117, value=false])
14:30:16.396 [main] DEBUG com.diozero.api.DigitalInputDevice.close - close()
14:30:16.403 [main] DEBUG com.diozero.internal.provider.AbstractDevice.close - close(), key=Native-GPIO-66
14:30:16.406 [main] DEBUG com.diozero.internal.provider.sysfs.SysFsDigitalInputDevice.closeDevice - closeDevice()
14:30:16.414 [main] ERROR com.diozero.internal.provider.AbstractDevice.close - Error closing device Native-GPIO-66: com.diozero.util.RuntimeIOException: java.io.IOException: Invalid argument: com.diozero.util.RuntimeIOException: java.io.IOException: Invalid argument
        at com.diozero.internal.provider.sysfs.SysFsDeviceFactory.unexport(SysFsDeviceFactory.java:225)
        at com.diozero.internal.provider.sysfs.SysFsDigitalInputDevice.closeDevice(SysFsDigitalInputDevice.java:134)
        at com.diozero.internal.provider.AbstractDevice.close(AbstractDevice.java:66)
        at com.diozero.api.DigitalInputDevice.close(DigitalInputDevice.java:99)
        at com.diozero.sampleapps.ButtonTest.test(ButtonTest.java:74)
        at com.diozero.sampleapps.ButtonTest.main(ButtonTest.java:64)
Caused by: java.io.IOException: Invalid argument
        at java.io.FileOutputStream.writeBytes(Native Method)
        at java.io.FileOutputStream.write(FileOutputStream.java:326)
        at sun.nio.cs.StreamEncoder.writeBytes(StreamEncoder.java:221)
        at sun.nio.cs.StreamEncoder.implClose(StreamEncoder.java:316)
        at sun.nio.cs.StreamEncoder.close(StreamEncoder.java:149)
        at java.io.OutputStreamWriter.close(OutputStreamWriter.java:233)
        at com.diozero.internal.provider.sysfs.SysFsDeviceFactory.unexport(SysFsDeviceFactory.java:224)
        at com.diozero.internal.provider.sysfs.SysFsDigitalInputDevice.closeDevice(SysFsDigitalInputDevice.java:134)
        at com.diozero.internal.provider.AbstractDevice.close(AbstractDevice.java:66)
        at com.diozero.api.DigitalInputDevice.close(DigitalInputDevice.java:99)
        at com.diozero.sampleapps.ButtonTest.test(ButtonTest.java:74)
        at com.diozero.sampleapps.ButtonTest.main(ButtonTest.java:64)
14:30:16.424 [main] DEBUG com.diozero.internal.DeviceStates.closed - closed(Native-GPIO-66)
14:30:16.432 [DIO-Zero Shutdown Handler] DEBUG com.diozero.util.ShutdownHandlerThread.run - Shutdown handler running
14:30:16.438 [diozero-EpollNative-waitForEvents-9635026] DEBUG com.diozero.util.EpollNative.waitForEvents - Finished
14:30:16.442 [diozero-EpollNative-processEvents-9635026] DEBUG com.diozero.util.EpollNative.processEvents - No event returned
14:30:16.446 [diozero-EpollNative-processEvents-9635026] DEBUG com.diozero.util.EpollNative.processEvents - Finished
14:30:16.452 [DIO-Zero Shutdown Handler] DEBUG com.diozero.util.DioZeroScheduler.shutdown - Shutdown - done
14:30:16.456 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.provider.AbstractDeviceFactory.close - close()
14:30:16.459 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.DeviceStates.closeAll - closeAll()
14:30:16.472 [DIO-Zero Shutdown Handler] DEBUG com.diozero.util.ShutdownHandlerThread.run - Shutdown handler finished

drogga, 2019-04-08:
I took a look at this issue. Simple fix, for some reason it does not like your try with resources statement in the unexport method of the com.diozero.internal.provider.sysfs.SysFsDeviceFactory. I changed it to an explicit close of the FileWriter and that resolved the issue. 

	void unexport(int gpio) {
		if (isExported(gpio)) {
			try {
				Writer unexport_writer = new FileWriter(rootPath.resolve(UNEXPORT_FILE).toFile());
				unexport_writer.write(String.valueOf(gpio));
				unexport_writer.close();
			} 
			catch (IOException e) {
				throw new RuntimeIOException(e);
			}
		}
	}

drogga, 2019-04-08:
Turns out this has to do with cape universal being enabled in more recent images. It exports all of the GPIOs and does not allow you to unexport them. Similar issue https://github.com/fivdi/onoff/issues/50
I added a line in the exception handler to consume it if there is an "Invalid argument" message

	void unexport(int gpio) {
		if (isExported(gpio)) {
			try (Writer unexport_writer = new FileWriter(rootPath.resolve(UNEXPORT_FILE).toFile())) {
				unexport_writer.write(String.valueOf(gpio));
			} catch (IOException e) {
				if(!e.getMessage().equalsIgnoreCase("Invalid argument"))
				throw new RuntimeIOException(e);
			}
		}
	}

mattjlewis, 2019-04-08:
Thanks for investigating. I will add that change and push a new snapshot build ASAP.

closed 2020-02-11 by mattjlewis