mattjlewis/diozero #28: Can't disable Log on MRFC522.isNewCardPresent()

Leo-Virgolini, 2018-01-13:
Logger prints to console : `"Timed out waiting for interrupt"` 
after calling **`RFID.isNewCardPresent()`**, even after setting **MFRC522.setLogReadsAndWrites(_false_);**

How do I disable it, or remove all Loggers completely?

mattjlewis, 2018-01-17:
Hi Leo, sorry for the delay replying. You need to override the default tinylog configuration. There are a number of ways to do this, see [tinylog configuration](http://www.tinylog.org/configuration). I'd suggest bundling your own tinylog.properties configuration file in your application.

If you're running my sample apps you should be able to specify your own config file via the -Dtinylog.configuration property. I'm struggling to get this to work however.

Regards,
Matt

mattjlewis, 2018-01-18:
I was being stupid by placing command line properties after the main class.
The best option would be:
`java -Dtinylog.configuration=yourtinylog.properties <main class name>`
Alternatively you can just change the log level for all classes in com.diozero:
`java -Dtinylog.level@com.diozero=warning <main class name>`
Matt

closed 2018-01-18 by mattjlewis