mattjlewis/diozero #29: DeviceFactoryHelper error

avysochin, 2018-01-17:
Tinker board have the error: DeviceFactoryHelper.init - Using native factory class **SysFsDeviceFactory** after adding Java Device I/O JAR to my local Maven repository, the error changes to another: DeviceFactoryHelper.init - Using native factory class **JdkDeviceIoDeviceFactory.**  Also there is LibraryLoader.loadlibrary - Loaded library 'diozero-system-utils' from classpath exception. GPIO_C_Library installed, and directly from the shell GPIO is working excellent.

mattjlewis, 2018-01-17:
Hi, could you be more specific on the error please.

Regards,
Matt

avysochin, 2018-01-18:
i think Java Device I/O and Diozero working separately from each other. How need to connect it ?

mattjlewis, 2018-01-18:
Question - why do you want to use JDK Device I/O? Try running with just diozero-core (and tinylog) first.
Regards,
Matt

avysochin, 2018-01-18:
How Diozero select in what board have been running? Rpi, BB or Tinker. i think Java Device I/O with wiringpi, pi4j is one of a providers for Diozero. Or not ?

mattjlewis, 2018-01-18:
diozero automatically detects the board it is running on. The builtin "sysfs" provider works on all boards that I list as supported. You don't actually need to use pi4j / wiringpi or any of the other providers.

Matt

avysochin, 2018-01-18:
DeviceFactoryHelper.init - Using native factory class SysFsDeviceFactory 
LibraryLoader.loadlibrary - Loaded library 'diozero-system-utils' from classpath 
are not error but only debug because maven run with -e flag

mattjlewis, 2018-01-19:
Please move this discussion to the [Google Group](https://groups.google.com/forum/#!forum/diozero) as I don't believe this is an issue with diozero.

closed 2018-01-19 by mattjlewis