mattjlewis/diozero #30: DeviceFactoryHelper

avysochin, 2018-01-17:
Why Maven could not find package com.diozero.util.DeviceFactoryHelper ?

mattjlewis, 2018-01-17:
How are you running this test - can you share the command line please.

avysochin, 2018-01-18:
mvn exec:java -Dexec.mainClass="tinkerexample.App",

mattjlewis, 2018-01-18:
Ah, ok. Please use just diozero-core and not diozero-provider-sysfs. diozero-provider-sysfs was merged into diozero-core.
See the following documentation page: [Creating Your Own diozero Application](http://www.diozero.com/en/latest/Application/).
Matt

avysochin, 2018-01-18:
Dear Matt

cut the pom - file
<dependency>
	    <groupId>com.diozero</groupId>
	    <artifactId>diozero-core</artifactId>
	    <version>0.10</version>
</dependency>

have delete sources files diozero/util/*.java from my local project directory 
the run
mvn compile

Error: cannot find symbol: class DeviceFactoryHelper
Location: package com.diozero.util

mattjlewis, 2018-01-19:
Glad to hear it is now working, let me know of any other issues.

closed 2018-01-19 by mattjlewis