mattjlewis/diozero #31: PCF8591 freezes RPi

z0rbas, 2018-01-18:
Hi,
I am using the following board: http://www.joy-it.net/en/explorer-700/
The example code of the board: http://anleitung.joy-it.net/wp-content/uploads/2016/12/Explorer700.zip

I have connected a photoresistor to analog A0 and can read the changes with the example code:

```
#include <wiringPi.h>
#include <pcf8591.h>
#include <stdio.h>

#define Address 0x48
#define BASE 64
#define A0 BASE+0
#define A1 BASE+1
#define A2 BASE+2
#define A3 BASE+3

int main()
{
	if (wiringPiSetup() < 0)return 1;
	pcf8591Setup(BASE,Address);

	while(1)
	{
  		printf("AD: %4dmv \n",analogRead(A0)*3300/255);
		delay(1000);
  	}
}
```

Since that worked I wanted to do the same in java. I encountered the problem that just calling:
`PCF8591 pcf8591 = new PCF8591();` freezes the RPi loosing ssh and http access to the wildfly swarm server. As dependencies I have diozero-core and diozero-provider-pi4j.

Does this problem sound familiar? I might be using the library in a wrong way.

I have tried debugging it, but have limited hardware experience (more a higher level dev).

mattjlewis, 2018-01-18:
Hi, first of all could you please try with just diozero-core and without diozero-provider-pi4j.

Matt

z0rbas, 2018-01-18:
Hi Matt,

Thank you for the answer and, wow that did the trick!
I must have misread the documentation. My train of thought was: ok pull in the core, and I'm using pi4j so get the right provider.
Can't remember how many times (a lot) I had to pull the plug because on every call RPi froze... Even compared C code from multiple locations with diozero code to find what exactly was different to make it die that way. Would be great if it bailed with some kind of message "You're doing it wrong!" if possible.

Out of curiosity, can you suspect what caused this behavior? (Other than me just pulling in the wrong dependency.)

Thanks again!

mattjlewis, 2018-01-18:
Glad it worked. The main ideas behind diozero is that it can work on any device plus presents a user logical and consitent API. The device independence was origanally delivered via the pluggable providers - on the Raspberry Pi you would use diozero-provider-pi4j (along with pi4j on the classpath). I then developed my own platform independent provider "sysfs" and included it within diozero core - this is the provider that I use most often.
Having to pull the plug on a Raspberry Pi is very annoying - lucky you've not corrupted the SD card.
I suspect the freeze that you are experiencing is related to the native code inside pi4j.
Best of luck and hope you enjoy the library.
Matt

closed 2018-01-18 by mattjlewis

z0rbas, 2018-01-21:
Ok. Thanks for the clarification.

I had already experimented with other sensors and pi4j alone but the code is nowhere as elegant as using diozero. I'll probably migrate those too to diozero if possible (and if I have enough time).

And of course, thank you very much for the lib!