mattjlewis/diozero #32: Rpi 3 - Unable to determine hardware version (BCM2835)

ricardojlrufino, 2018-03-03:
How solve this?
```
Unable to determine hardware version. I see: Hardware	: BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.
```

Kernel:
Linux raspberrypi 4.14.22-v7+ #1096 SMP Mon Feb 26 19:14:22 GMT 2018 armv7l GNU/Linux

ricardojlrufino, 2018-03-04:
```
        <dependency>
            <groupId>com.diozero</groupId>
            <artifactId>diozero-provider-pi4j</artifactId>
            <version>0.10</version>
        </dependency>

        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-core</artifactId>
            <version>1.1</version>
        </dependency>
```

Using PI4J direct API it's working ...

mattjlewis, 2018-03-04:
I believe this is a problem with pi4j itself relating to it containing a C library this is statically linked to an old version of wiringPi. There is a command line option to force it to use the dynamically linked version (Google should be able to tell you the exact command line option).
Let me know how you get on with that.
I've not yet tried pi4j 1.2-SNAPSHOT.
If you are using diozero I would recommend using the built-in provider (only include to diozero-core and tinylog). Is there a reason you need to use the pi4j provider?
Matt

ricardojlrufino, 2018-03-05:
I was trying to use the RFID reader. I thought it would work only using a pi4j provider. After your comment I used built-in provider and it worked

I tested using the pi4j API directly (1.2-SNAPSHOT) and it works correctly. When I use pi4j provider + pi4j 1.2-SNAPSHOT, a strange file system error occurs and hangs the raspberry

I was in doubt of which is the advantage of pi4j provider ?

ricardojlrufino, 2018-03-05:
PS: Actually the problem is in the pi4j provider, I tried to use a PWM function and the same error happened.

mattjlewis, 2018-03-05:
Great. Pi4j support in diozero largely historic; the portability of the library is based on the pluggable providers. Support via pi4j was easy and added before I developed my own portable provider that will work on any device. If you require high performance GPIO control on the Pi you would need to use pigpio or my own memory mapped implementation.
If I get a moment I will look into the pi4j problem myself. Did you find the command line flag to force loading the dynamic pi4j / wiringPi library?

ricardojlrufino, 2018-03-05:
"Unable to determine hardware version. I see: Hardware	: BCM2835", this has been solved using 1.2-SNAPSHOT, no need flag (-Dpi4j.linking=dynamic).

I got another error: https://github.com/Pi4J/pi4j/issues/319#issuecomment-370280325
But i'll try (PWM / SPI) as root ... when I'm home

mattjlewis, 2018-03-05:
Do you have a reason for using pi4j? You might want to try the pigpio provider, I've found it to be quite reliable and has excellent PWM support (not just on the Pi's hardware PWM pins). It does, however, also require you to run as root.

ricardojlrufino, 2018-03-05:
> Do you have a reason for using pi4j?
I believe it is the best known, and would have better support

thanks for assist

I'm planning on integrating your library as the basis of my framework when it comes to SBC
http://opendevice.io

mattjlewis, 2018-05-14:
Great, let me know how you get on.

closed 2018-05-14 by mattjlewis