mattjlewis/diozero #33: Releasing 0.11 version

jakekarnes42, 2018-03-26:
Hi Matt, 

Thanks for the great library! I'd love to use the latest version of the library in a project I'm working on. I see the most recent release on Maven Central is 0.10, and I've seen in the README that the 0.11-SNAPSHOT jars are available via Google Drive. 

I was just curious when you're planning to release 0.11 to Maven so it can be used in a project more easily. 

Thanks again!

mattjlewis, 2018-03-28:
Hi Jake - good point, 0.11 snapshot is very stable. I will release this weekend.

jakekarnes42, 2018-04-02:
Hey Matt, any chance the release could be made soon? Thank you for spending time on this.

mattjlewis, 2018-04-02:
Sorry - will do tomorrow am!

jakekarnes42, 2018-04-03:
Thank you very much!