mattjlewis/diozero #34: SpiDevice constructor

avysochin, 2018-04-27:
In attempts to create SpiDevice with particular CLOCK_FREQUENCY and MODE
1) foo = new SpiDevice(2,0, DEFAULT_SPI_CLOCK_FREQUENCY, DEFAULT_SPI_CLOCK_MODE, DEFAULT_LSB_FIRST);
2) foo = new SpiDevice(2, 0, 1000000, SpiClockMode.MODE_0, DEFAULT_LSB_FIRST);
3) foo = new SpiDevice(2,0);
works only 3th way
[ERROR] /src/main/java/App.java:[72,30] error: cannot find symbol
[ERROR] symbol:   variable DEFAULT_SPI_CLOCK_FREQUENCY
[ERROR] location: class App
[ERROR] /src/main/java/App.java:[72,59] error: cannot find symbol
[ERROR] symbol:   variable DEFAULT_SPI_CLOCK_MODE
[ERROR] location: class App
[ERROR] /src/main/java/App.java:[72,83] error: cannot find symbol

avysochin, 2018-04-28:
Made the changes:
SpiClockMode.MODE_0 changed to SpiClockMode.getByMode(0)
DEFAULT_LSB_FIRST changed to true
compiling is ok, but runtime errors:

DEBUG com.diozero.util.DeviceFactoryHelper.init - Using native device factory class SysFsDeviceFactory
DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
DEBUG com.diozero.internal.provider.sysfs.NativeSpiDevice.<init> - Opening /dev/spidev2.0, frequency 1000000 Hz, mode MODE_0

com.diozero.util.RuntimeIOException: Error in spiTransfer(), response: -1
	at com.diozero.internal.provider.sysfs.NativeSpiDevice.writeAndRead(NativeSpiDevice.java:109)
	at com.diozero.internal.provider.sysfs.NativeSpiDevice.writeAndRead(NativeSpiDevice.java:102)
	at com.diozero.internal.provider.sysfs.SysFsSpiDevice.writeAndRead(SysFsSpiDevice.java:81)
	at com.diozero.api.SpiDevice.writeAndRead(SpiDevice.java:106)

Bad file descriptorSPI message transfer failed

mattjlewis, 2018-05-14:
Looks like a coding error on your behalf?

This compiles for me:
``` 
public class SpiApp {
	public static void main(String[] args) {
		SpiDevice foo = new SpiDevice(2, 0, SPIConstants.DEFAULT_SPI_CLOCK_FREQUENCY,
				SPIConstants.DEFAULT_SPI_CLOCK_MODE, SPIConstants.DEFAULT_LSB_FIRST);
		foo = new SpiDevice(2, 0, 1000000, SpiClockMode.MODE_0, SPIConstants.DEFAULT_LSB_FIRST);
		foo = new SpiDevice(2, 0);
	}
}

```

closed 2018-05-14 by mattjlewis