mattjlewis/diozero #35: diozero-ws281x-java Not working with RPI 3B+

petercinibulk, 2018-05-20:
Getting a runtime error thrown when trying to initialize a ws281x led strip saying hardware is not supported.

Works perfectly if I swap sd-card into a RPI 3B

petercinibulk, 2018-05-21:
ws2811_init failed: Hardware revision is not supported
java.lang.RuntimeException: Error initialising the WS281x strip
        at com.diozero.ws281xj.rpiws281x.WS281x.<init>(WS281x.java:134)
        at com.diozero.ws281xj.rpiws281x.WS281x.<init>(WS281x.java:102)

petercinibulk, 2018-05-23:
<dependency>
            <groupId>com.diozero</groupId>
            <artifactId>diozero-ws281x-java</artifactId>
            <version>0.11</version>
</dependency>

I believe this is the latest version. 
The sample apps do not work either.

petercinibulk, 2018-05-29:
Any update on this?

mattjlewis, 2018-05-29:
That is indeed the latest version of diozero, however, you need to look at the version of the rpi_ws281x C library which is outside of diozero. As I said, the error message comes from the rpi_ws281x library. You can always raise an issue on https://github.com/jgarff/rpi_ws281x
Matt

petercinibulk, 2018-05-30:
I'm a little confused. I did not install the rpi_ws281x C library and it worked on the rpi 3b. I just installed it with the latest build (they verified that it works with the 3b+) but I still got the same error on the 3b+. Does the C library need to be in the same directory as the jar file?

mattjlewis, 2018-06-26:
The C library should be present on the Raspberry Pi as part of the base Raspbian install. I will check

HoggenMari, 2019-03-08:
Are all the jar-files of the 0.12-SNAPSHOT available to download? I have problems at the moment to run the WS281xTest example with the RaspberryPi 3: I compiled the latest diozero-ws281x-java, but I get the following error message "Error: java.lang.UnsatisfiedLinkError: com.diozero.ws281xj.rpiws281x.WS281xNative.initialise(IIIIIII)Ljava/nio/ByteBuffer;"

e-fadil, 2019-05-11:
Any news on this? I am getting the same issue as HoggenMari

mattjlewis, 2019-05-12:
I reproduced the error on my Pi, rebuilt the native library and committed changes into 0.12-SNAPSHOT. Will also push a new release on GitHub as soon as the snapshot deploy finishes.

closed 2019-05-12 by mattjlewis