mattjlewis/diozero #36: Serial bit banging?

9903286, 2018-05-21:
Hello, would it be possible to add pigpio's serial bit-banging to this library?

mattjlewis, 2018-05-22:
To be clear, are you referring to the 4 bb_serial_* methods as described in the Advanced section here: http://abyz.me.uk/rpi/pigpio/pdif2.html?

9903286, 2018-05-22:
Yes, and also pretty much everything in the WAVES section, as bb_serial_* are used for receiving while the wave_* are used for transmitting. This answer on stackexchange shows the neccessery stuff for a bit-banged serial port: https://raspberrypi.stackexchange.com/a/27490

mattjlewis, 2018-05-29:
Apologies, but that looks like quite a bit of effort; I'm quite busy with work at the moment. You are welcome to contribute support (to my pigpioj library). https://github.com/mattjlewis/pigpioj
Matt

9903286, 2018-05-30:
Huh, i'm actually making progress with this, might take a bit more time, but so far i'm able to send on software serial and i'm getting the number of bytes that i've received on software serial.

closed 2020-02-11 by mattjlewis