mattjlewis/diozero #39: JVM crash on exit when using WS2812 on RPi Zero W

Zappes, 2018-08-05:
I've just started writing an application for the RPi Zero W that uses a WS2812 strip. When creating my rudimentary service, I noticed that the JVM reproducibly crashes when the WS281x is closed.

I have attached the error log created by the JVM: [hs_err_pid909.log](https://github.com/mattjlewis/diozero/files/2260091/hs_err_pid909.log)

The code that caused this crash on exit can be found here: https://github.com/Zappes/Henkinson/blob/master/src/main/java/net/bluephod/henkinson/Henkinson.java (commit 2f46b54)

Great library BTW - I really like it a lot!

mattjlewis, 2018-12-30:
Thanks and apologies for the very (!) long delay replying. Does this crash still happen? I've just committed a new snapshot build (0.12-SNAPSHOT) that includes an updated version of the ripws281x native library. BTW, I also implemented a pure Java version that uses SPI.

mattjlewis, 2021-01-16:
This should now be fixed.

closed 2021-01-16 by mattjlewis