mattjlewis/diozero #40: Forked repo, can't build because of missing dependencies

Zappes, 2018-08-05:
I just forked the repo as I consider contributing to the code base. When I try to build with `mvn clean install`, the build fails because jdk.dio:device-io:jar:1.0 can't be found. A log of the build is attached: 
[build.log](https://github.com/mattjlewis/diozero/files/2260096/build.log)

Do I have to do something specific prior to building the project? If that should be the case, it would be extremely helpful if a little help was provided in the README.

Zappes, 2018-08-05:
A closer look shows that some more dependencies can't be found in Maven Central:

[image]

Sergeant007, 2018-08-09:
Same with me. I have few ideas with code improvements, but postpone them, since I cannot build the project.

Sergeant007, 2018-08-25:
Is this possible just to move JDK providers to a separate GitHub repo? Documentation says:
> In theory the OpenJDK Device I/O service provider should provide the best platform support, however, the JDK Device I/O library doesn't support PWM. 

Absence of software PWM, publicly accessible Maven artifact and slow development activity (one commit per 2 months to http://hg.openjdk.java.net/dio/dev) could probably mean that OpenJDK Device I/O is not mature enough to be used.

mattjlewis, 2019-03-25:
Resolved in 0.12-SNAPSHOT

closed 2019-03-25 by mattjlewis