mattjlewis/diozero #43: UnsatisfiedLinkError using bbbio provider

drogga, 2019-03-24:
I unzip the file diozero-distribution-0.11-SNAPSHOT-bin.zip downloaded from the google drive link on a BBB running the latest version of debian. I move to the folder and execute the command:

java -classpath tinylog-1.2.jar:diozero-core-0.11-SNAPSHOT.jar:diozero-sampleapps-0.11-SNAPSHOT.jar:diozero-provider-bbbiolib-0.11-SNAPSHOT.jar com.diozero.sampleapps.ButtonTest 66

I get the error below. My understanding is that BBBio is included in the corresponding jar file and there is no need to do anything else to get this to work (the source seems to indicate this). I get this error running 0.11 or 0.10 
Also, I can get the sysfs version working on the BBB, it can flash an LED but it will not read input from a switch, switch is connected to GPIO and GNDD and I am running the ButtonTest sample app.
Any help is appreciated

# java -classpath tinylog-1.2.jar:diozero-core-0.11-SNAPSHOT.jar:diozero-sampleapps-0.11-SNAPSHOT.jar:diozero-provider-bbbiolib-0.11-20171009.203154-13.jar com.diozero.sampleapps.ButtonTest 66

Exception in thread "main" java.util.ServiceConfigurationError: com.diozero.internal.provider.NativeDeviceFactoryInterface: Provider com.diozero.internal.provider.bbbiolib.BbbIoLibDeviceFactory could not be instantiated
        at java.util.ServiceLoader.fail(ServiceLoader.java:232)
        at java.util.ServiceLoader.access$100(ServiceLoader.java:185)
        at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:384)
        at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
        at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
        at java.util.Spliterators$IteratorSpliterator.tryAdvance(Spliterators.java:1812)
        at java.util.stream.ReferencePipeline.forEachWithCancel(ReferencePipeline.java:126)
        at java.util.stream.AbstractPipeline.copyIntoWithCancel(AbstractPipeline.java:498)
        at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:485)
        at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471)
        at java.util.stream.FindOps$FindOp.evaluateSequential(FindOps.java:152)
        at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
        at java.util.stream.ReferencePipeline.findFirst(ReferencePipeline.java:464)
        at com.diozero.util.DeviceFactoryHelper.init(DeviceFactoryHelper.java:69)
        at com.diozero.util.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:88)
        at com.diozero.api.DigitalInputDevice.<init>(DigitalInputDevice.java:71)
        at com.diozero.devices.Button.<init>(Button.java:80)
        at com.diozero.sampleapps.ButtonTest.test(ButtonTest.java:68)
        at com.diozero.sampleapps.ButtonTest.main(ButtonTest.java:64)
Caused by: java.util.ServiceConfigurationError: com.diozero.internal.provider.NativeDeviceFactoryInterface: Provider com.diozero.internal.provider.bbbiolib.BbbIoLibDeviceFactory could not be instantiated
        at java.util.ServiceLoader.fail(ServiceLoader.java:232)
        at java.util.ServiceLoader.access$100(ServiceLoader.java:185)
        at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:384)
        at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
        at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
        at java.util.Spliterators$IteratorSpliterator.tryAdvance(Spliterators.java:1812)
        at java.util.stream.ReferencePipeline.forEachWithCancel(ReferencePipeline.java:126)
        at java.util.stream.AbstractPipeline.copyIntoWithCancel(AbstractPipeline.java:498)
        at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:485)
        at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471)
        at java.util.stream.FindOps$FindOp.evaluateSequential(FindOps.java:152)
        at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
        at java.util.stream.ReferencePipeline.findFirst(ReferencePipeline.java:464)
        at com.diozero.util.DeviceFactoryHelper.init(DeviceFactoryHelper.java:69)
        at com.diozero.util.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:88)
        at com.diozero.util.LibraryLoader.loadLibrary(LibraryLoader.java:53)
        at com.diozero.internal.provider.bbbiolib.BbbIoLibNative.<clinit>(BbbIoLibNative.java:39)
        at com.diozero.internal.provider.bbbiolib.BbbIoLibDeviceFactory.<init>(BbbIoLibDeviceFactory.java:48)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at java.lang.Class.newInstance(Class.java:442)
        at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:380)
        ... 16 more
Caused by: java.lang.UnsatisfiedLinkError: com.diozero.internal.provider.bbbiolib.BbbIoLibNative.init()I
        at com.diozero.internal.provider.bbbiolib.BbbIoLibNative.init(Native Method)
        at com.diozero.internal.provider.bbbiolib.BbbIoLibDeviceFactory.<init>(BbbIoLibDeviceFactory.java:48)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at java.lang.Class.newInstance(Class.java:442)
        at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:380)
        ... 37 more

mattjlewis, 2019-03-25:
Could you please try with the latest snapshot (just pushed to maven-central and also uploaded to Google Drive). I will need to dig out my BBB and give it a go myself when I get a moment. The unsatisfied link error suggests it cannot load the shared library, could be related to the BBBio library itself.
BBBio: https://github.com/tomtzook/BBBIO
Google Drive link: https://drive.google.com/open?id=1ojGkcxQLT_y1wPCQWdG4HQcsnKY55Me8

drogga, 2019-03-27:
Thank you for the update, unfortunately it still has the same issue.
I tried stepping through the code it gets to the point of attempting to load BBBIO in the classloader and then fails for some unknown reason.

I am also experiencing issue #27 and commented on it (they are not related)

mattjlewis, 2019-04-09:
Looking into this now, I've managed to get the library loaded successfully using a bit of a hack. Just looking for a proper fix.

mattjlewis, 2019-10-12:
FYI, this should now fixed in the latest 0.12 snapshot build.

closed 2020-02-11 by mattjlewis