mattjlewis/diozero #44: Support for MRAA library

TheNetStriker, 2019-04-08:
Would it be possible to add a provider for the Intel MRAA library? (https://github.com/intel-iot-devkit/mraa)

bootsy52, 2019-07-23:
This is exactly what I wanted to ask, too. 

Because the Rock Pi 4 made by radxa have done a port for mraa instead of WiringPi. The concept behing MRAA looks good. Accessing the mem directly would require diozero to run as root, right? So this is something i would avoid if possible

mattjlewis, 2019-07-27:
Hi, this looks like it should be fairly straightforward, I will take a look.
First of all, I'd like to understand why - from what I can see, the mraa library provides similar functionality to diozero - I have developed a similar and very minimal C interface that is very easy to port to other hardware platforms for the functions that cannot be performed in Java (ioctl commands mainly).
Yes, accessing /dev/mem directly does require root permissions - this is typically only required for very fast GPIO switching.

closed 2020-02-11 by mattjlewis