mattjlewis/diozero #45: mfrc522: ReadAndWrite Example

giuseppCl, 2019-04-09:
Hi there, 

i'm using your mfrc522 sampleapps to learn a bit about  the Reader/Writer. 
I can successfully read data from secotr's/blocks but i'm not able to write data.

**My changes:**

Changed boolean from false to true. 
`readAndWrite(rfid, uid, auth_key, true);`

I choosed sector 6 
```
                byte blocks_per_sector = 4;
		byte sector = 6;
		byte blockAddr = (byte) (sector * blocks_per_sector);
		byte trailerBlock = (byte) (blockAddr + (blocks_per_sector - 1));
```

and my datablock is

```
byte dataBlock[] = new byte[] {
					0x05, 0x05, 0x05, 0x05,			
					0x05, 0x05, 0x05, 0x05,			
					0x05, 0x05, 0x05, 0x05,			
					0x05, 0x05, 0x05, 0x05			
			};
	
```

**This is the error message:**

> 19:08:04.682 [main] INFO com.diozero.sampleapps.mfrc522.ReadAndWrite.readAndWrite - Writing data into block 0x18 : 0x05050505 05050505 05050505 05050505 ...
**19:08:04.683 [main] ERROR com.diozero.sampleapps.mfrc522.ReadAndWrite.readAndWrite - MIFARE_Write() failed: INVALID**
19:08:04.685 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - FIFO Level: 1
19:08:04.686 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - ok
19:08:04.688 [main] DEBUG com.diozero.api.SpiDevice.close - close()
19:08:04.689 [main] DEBUG com.diozero.internal.provider.AbstractDevice.close - close(), key=Native-SPI-8-0
19:08:04.690 [main] DEBUG com.diozero.internal.DeviceStates.closed - closed(Native-SPI-8-0)
19:08:04.697 [DIO-Zero Shutdown Handler] DEBUG com.diozero.util.ShutdownHandlerThread.run - Shutdown handler running
19:08:04.702 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.provider.AbstractDeviceFactory.close - close()
19:08:04.703 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.DeviceStates.closeAll - closeAll()
19:08:04.712 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.provider.AbstractDevice.close - close(), key=Native-GPIO-25
19:08:04.713 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.provider.pigpioj.PigpioJDigitalOutputDevice.closeDevice - closeDevice()
19:08:04.713 [DIO-Zero Shutdown Handler] DEBUG com.diozero.internal.DeviceStates.closed - closed(Native-GPIO-25)
19:08:04.729 [DIO-Zero Shutdown Handler] DEBUG com.diozero.util.ShutdownHandlerThread.run - Shutdown handler finished

I tried to figure out on my own. So i looked up the implementation of the method:

`status = rfid.mifareWrite(blockAddr, dataBlock);
`
```
public StatusCode mifareWrite(byte blockAddr, byte[] buffer) {
		// Sanity check
		if (buffer == null || buffer.length != 16) {
			return StatusCode.INVALID;
		}
		
		// Mifare Classic protocol requires two communications to perform a write.
		// Step 1: Tell the PICC we want to write to block blockAddr.
		byte[] cmdBuffer = { PiccCommand.MF_WRITE.getValue(), blockAddr };
		StatusCode result = mifareTransceive(cmdBuffer); // Adds CRC_A and checks that the response is MF_ACK.
		if (result != StatusCode.OK) {
			return result;
		}
		
		// Step 2: Transfer the data
		return mifareTransceive(buffer); // Adds CRC_A and checks that the response is MF_ACK.
	} // End MIFARE_Write()
```
I think this method throws SatusCode Invalid when the byte Array is Null or whether its length is not equal 16. But thats not the case in my Code. 

thank you in advance

jimontoyag, 2019-04-27:
It seems the problem is in the method 
`public StatusCode mifareTransceive(byte[] sendData, boolean acceptTimeout) {
		// Sanity check
		if (sendData == null || sendData.length != 16) {
			return StatusCode.INVALID;
		}`
Because in the mifareWrite he is sending this array `byte[] cmdBuffer = { PiccCommand.MF_WRITE.getValue(), blockAddr };` to the sendData parameter.
Maybe replace that block with communicateWithPICC to send the MF_READ instruction?

giuseppCl, 2019-04-29:
Hi, thank you for your advice.
I solved this problem by commenting out 

`
	// Sanity check
		if (buffer == null || buffer.length != 16) {
			return StatusCode.INVALID;
		}
`
Due to that i'm able to change the access bits in the sector trailer.

mattjlewis, 2019-04-29:
Excellent, thank you. I will incorporate into the other fixes that I have been making recently.

closed 2019-07-27 by mattjlewis