mattjlewis/diozero #46: WS281x on raspberry PI 4 hardware revision.

e-fadil, 2019-07-01:
Hi, I'm trying to use the WS281x library on raspberry pi 4 but get the error below, works fine on rasberry pi 3.

JDK Server VM warning: You have loaded library /tmp/libws281xj10313733273385733761.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Error loading library from classpath, trying from system library path
Loaded library 'ws281xj' from system library path
ws2811_init failed: Hardware revision is not supported
Exception in thread "main" java.lang.RuntimeException: Error initialising the WS281x strip
	at com.diozero.ws281xj.WS281x.<init>(WS281x.java:165)
	at com.diozero.ws281xj.WS281x.<init>(WS281x.java:132)
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x9351e664, pid=1036, tid=1058
#
# JRE version: OpenJDK Runtime Environment (11.0.3+7) (build 11.0.3+7-post-Raspbian-5)
# Java VM: OpenJDK Server VM (11.0.3+7-post-Raspbian-5, mixed mode, g1 gc, linux-)
# Problematic frame:
# C  [libws281xj.so+0x2664]  ws2811_fini+0xc
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/pi/Desktop/hs_err_pid1036.log
#
# If you would like to submit a bug report, please visit:
#   Unknown
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

mattjlewis, 2019-07-27:
I now have a Raspberry Pi 4 so will take a look. Should simply be a recompile of the library.

mattjlewis, 2019-10-12:
This should now work with the latest snapshot build. Please let me know so that I can close the issue.

e-fadil, 2019-10-12:
Great. The latest snapshot build uploaded still seems to 0.12-SNAPSHOT3 .. is it somewhere else?

mattjlewis, 2019-10-13:
It should be available in the sonatype snapshot repositories. I've also uploaded it here:
https://drive.google.com/open?id=1RjUc4_ko6vhjSKHYhV4zuyAhjY3_yFv7

This page gives instructions on setting up your development environment, including instructions on using the diozero snapshot builds.

e-fadil, 2019-10-23:
Thanks!. that works.

Tortoc, 2020-04-11:
Got that issue today with an Raspberry Pi 4 rev. 1.2 on version 0.12-SNAPSHOT.
Executing the test in jgarff-folder works. 

Is there a newer version for rev. 1.2 support?

mattjlewis, 2020-04-11:
I’ve checked in a new build into GitHub. I’ll publish a new snapshot build soon.

Regards,
Matt

mattjlewis, 2021-01-16:
Should now be fixed.

closed 2021-01-16 by mattjlewis