mattjlewis/diozero #48: Native jvm crash in libws281xj.so on exit on 3 B+

jmihalich, 2019-08-20:
Hi, I'm on the Pi 3 B+.  I'm using the 0.12 snapshot, with the exception that i pulled your master branch and recompiled the diozero-ws281x-java project to get the native dependency fix you checked in.  

At this point, everything works, i can light up the strip however i want, but when my code exits the jvm crashes in the native library.  I've attached the jvm dump.

[hs_err_pid1668.log](https://github.com/mattjlewis/diozero/files/3518768/hs_err_pid1668.log)

Let me know what other info you need to debug.

Thanks,
Joe

jmihalich, 2019-08-20:
It looks like it's crashing in the shutdown hook, from the stack traces in the log file.

mattjlewis, 2019-10-12:
Hi Joe, could try without pi4j and with the latest diozero snapshot build please. The terminate function was being called twice - you are correct that the crash was due to terminate being called for a second time by the shutdown hook.

I just ran my test application with latest code on a Pi3B and it appears to work ok:
```
java -cp tinylog-1.3.6.jar:diozero-core-0.12-SNAPSHOT.jar:diozero-ws281x-java-0.12-SNAPSHOT.jar com.diozero.ws281xj.sampleapps.WS281xTest 
Using GPIO 10
Loaded library 'ws281xj' from classpath
order=BIG_ENDIAN
rainbowColours()
test2()
Adding red...
Adding green...
Adding blue...
hsbTest()
hslTest()
colourWipe - red
colourWipe - green
colourWipe - blue
theatreChase - white
theatreChase - red
theatreChase - blue
rainbow
rainbowCycle
close()
```

Thank you for your patience.

Matt

jmihalich, 2019-10-22:
Hey, thanks.  That all worked fine.  The instructions on the github site said you needed one of the runtime libraries like pi4j so that's why i was using it.  Does your stuff not need that anymore?

Joe

mattjlewis, 2021-02-26:
No - I've removed support for pi4j as a backend in the latest release.

closed 2021-02-26 by mattjlewis