mattjlewis/diozero #49: Add support for x86 Linux

452, 2019-09-26:
Java dio for x86 support
I want to use Java and Groovy script for development, debugging, investigating my home automation.
But existing libraries work only for embedded boards like rpi.
I investigate and found some solutions like enhance i2c and GPIO on my laptop or stationary pc over USB adapter
https://github.com/harbaum/I2C-Tiny-USB
https://github.com/harbaum/I2C-Tiny-USB/tree/master/digispark

https://youtu.be/rriKd9jkdJQ

Digispark Attiny85 I2C-Tiny-USB
FT311D USB To I2C IIC SPI UART GPIO PWM Converter Adapter 3.3V/5V
FT232H Multifunction High-Speed USB to JTAG UART/ FIFO SPI/ I2C

https://github.com/dvlopt/linux-i2c.java
https://github.com/dvlopt/linux-gpio.java

mattjlewis, 2019-10-12:
Hi, I'm not sure what you are requesting here. You can use diozero for remotely connecting to a range of boards, both wired and wireless. You can connect to Raspberry Pis remotely using eight pigpio or my own remote protocol, you can connect to ESP type boards via USB and some via Bluetooth / Wifi.
Matt

452, 2020-02-11:
just how to use your library with laptop hardware with i2c bus? (I not need any embedded pc, rpi, etc)

I can extend numbers of i2c busses on my laptop with Linux OS, and use for development directly from Java

mattjlewis, 2020-02-11:
Interesting. The first thing would be to get the system-utils-native C project to compile on your laptop. Once you have that compiled you can then try using com.diozero.internal.provider.sysfs.NativeI2C. If all of this works, I can help you define a new board so that you can use the higher level abstractions.

mattjlewis, 2021-02-26:
Closing as no response.

closed 2021-02-26 by mattjlewis