mattjlewis/diozero #54: I2C read limit

michieltjampens, 2020-05-15:
Am i doing something wrong or is there a 32 byte limit to the amount of bytes that can be read in a single transfer? Or is this hardware related (running on beaglebone black) because there's no limit mentioned in the docs. I know it's not the chip that stops sending because  a logic analyser schows a stop after 32 bytes.

mattjlewis, 2020-05-27:
Hi, I'm not aware of any limit.

michieltjampens, 2020-05-27:
Hm, then i'm doing something wrong. For now i worked around it by dividing read instructions, but gues i'll have to look if the beaglebone restricts this somehow.

mattjlewis, 2020-05-27:
I'm not ruling out any issues with my library btw, just not sure if I've tested reads > 32 bytes. Let me know if there is anything to fix. Are you using the built-in provider or bbbiolib? Can you test with both?

michieltjampens, 2020-05-27:
Using the built-in provider, bbbiolib kept complaining about pwm so i would first need to fix that issue. But given that SPI and all the other I2C stuff works with the built-in this wasn't a priority to look into. 

I discovered the limit when trying to interface with a AT24C02D and checked the bus with a Saleae datalogger to be sure it's not on the AT's end.

mattjlewis, 2021-02-26:
Ok, I managed to get time to look into this. diozero mainly uses the [SMBus interface](https://www.kernel.org/doc/Documentation/i2c/smbus-protocol) for I2C transactions. This imposes a 32-byte limit on I2C reads and writes. I've recently updated my Javadoc in my [I2CSMBusInterface](https://github.com/mattjlewis/diozero/blob/master/diozero-core/src/main/java/com/diozero/api/I2CSMBusInterface.java) to make this clear.
As you'll see in the Linux Kernel documentation, there are two additional methods that do not have this constraint - i2c_smbus_read_i2c_block_data() and i2c_smbus_write_i2c_block_data(). I have [replicated these methods](https://github.com/mattjlewis/diozero/blob/master/diozero-core/src/main/java/com/diozero/api/I2CSMBusInterface.java#L343) within diozero as well as [two of my own extension methods](https://github.com/mattjlewis/diozero/blob/master/diozero-core/src/main/java/com/diozero/api/I2CSMBusInterface.java#L395).
I am also prototyping a "raw" I2C read-write interface, here is the relevant interface https://github.com/mattjlewis/diozero/blob/master/diozero-core/src/main/java/com/diozero/api/I2CDeviceInterface.java#L43.

michieltjampens, 2021-02-27:
Thanks, i'll look into those when i'm working with i2c memory again. 

Fyi i'm using Diozero to handle the i2c (and spi but not that extensive yet) communication for a datalogger. 
The datalogger itself is obviously written in java but all i2c communication is setup in xml (so pretty much script to java).

closed 2021-03-22 by mattjlewis