mattjlewis/diozero #55: MFRC522 on Raspberry Pi 3B+: Timed out waiting for interrupt

PauMAVA, 2020-05-18:
Hello!

I am trying to use an RFID MFRC522 module on a Raspberry PI 3B+. Once having enabled the SPI interface on the board, I am trying to use the following code in order to read the UID of a card (it is almost as the example one):

```java
private void listen() {
    log("Started listening for RFID cards...");
    try (MFRC522 mfrc522 = new MFRC522(0, 1, 22)) {
        while (listen) {
            rfidLoop(mfrc522);
            SleepUtil.sleepSeconds(1);
        }
    }
}

private void rfidLoop(MFRC522 rfid) {
    if (!rfid.isNewCardPresent()) {
        log("No card found.");
        return;
    }
    log("Card found!");
    UID uid = rfid.readCardSerial();
    if (uid == null) {
        log("Failed to read UID");
        return;
    } else {
    log("Read UID: " + ConversionTools.bytesToHex(uid.getUidBytes()) + "of type " + uid.getType().getName());
    }
    rfid.haltA();
    rfid.stopCrypto1();
}

static void log(String message) {
    MainClass.log(MainClass.ANSI_YELLOW + "RFID Sensor >> " + MainClass.ANSI_RESET + message);
}
``` 
The thing is that a UID is never read and the only output I get is: `Timed out waiting for interrupt`. Here you have an extract of the console:
```
20:21:58.766 [main] DEBUG com.diozero.devices.MFRC522.init - reset pin was off
20:21:59.039 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - Timed out waiting for interrupt
[19.428] RFID Sensor >> No card found.
20:22:00.255 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - Timed out waiting for interrupt
[20.631] RFID Sensor >> No card found.
20:22:01.458 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - Timed out waiting for interrupt
[21.833] RFID Sensor >> No card found.
20:22:02.661 [main] DEBUG com.diozero.devices.MFRC522.communicateWithPICC - Timed out waiting for interrupt
...
```
I have already checked the following:
- The SPI interface is enabled on `/boot/config.txt`
- Have checked the wiring:
  ([https://es.pinout.xyz/#](https://es.pinout.xyz/#)):
  * Pin 22 (GPIO 25): Reset
  * Pin 1: 3v3
  * Pin 9: Ground
  * Pin 21 (GPIO9): MISO
  * Pin 19 (GPIO 10): MOSI
  * Pin 23 (GPIO11): SCK
  * Pin 24 (GPIO 8): SDA
  
- Tried using chip select 0 instead of 1.
- Tried passing 25 instead of 22 for the reset pin parameter. (Tried this for both chip selects).

Here you also have the output of some related commands for reference:
```
$ ls -lah /dev/spi*
crw-rw---- 1 root spi 153, 0 May 17 21:17 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 May 17 21:17 /dev/spidev0.1
```
```
$ uname -a
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
```
```
$ cat /boot/config.txt
dtparam=spi=on
```
I can't find where is the problem. Is even my program being able to communicate with the sensor?

Any help is really appreciated.

PauMAVA, 2020-05-23:
@mattjlewis Any idea why is this happening?

PauMAVA, 2020-06-05:
It seems that a bad cable was causing the issue.

The code I ended up  using is:
```java
private void listen() {
    log("Started listening for RFID cards...");
    try (MFRC522 mfrc522 = new MFRC522(0, 0, 22)) {
        while (listen) {
            rfidLoop(mfrc522);
            SleepUtil.sleepSeconds(1);
        }
    }
}

private void rfidLoop(MFRC522 rfid) {
    if (!rfid.isNewCardPresent()) {
        log("No card found.");
        return;
    }
    log("Card found!");
    UID uid = rfid.readCardSerial();
    if (uid == null) {
        log("Failed to read UID");
        return;
    } else {
    log("Read UID: " + ConversionTools.bytesToHex(uid.getUidBytes()) + "of type " + uid.getType().getName());
    }
    rfid.haltA();
    rfid.stopCrypto1();
}

static void log(String message) {
    MainClass.log(MainClass.ANSI_YELLOW + "RFID Sensor >> " + MainClass.ANSI_RESET + message);
}

```

closed 2020-06-05 by PauMAVA

mattjlewis, 2020-06-19:
Excellent, good to know. The MFRC522 code is pretty complex - I ported Arduino code to Java and TBH I've not tested it that much. I've also experienced a bad cable - I was pulling my hair out tying to work out why something wasn't working that had been working.