mattjlewis/diozero #56: How to disable log DEBUG messages

PauMAVA, 2020-06-18:
Does this library support disabling debug messages? I have a finished project but I want to keep my logs clean. If debug messages are shown then my console gets filled with 7 messages or more every second.

Is there any parameter that allows to do this?

mattjlewis, 2020-06-18:
The library uses [tinylog](https://tinylog.org/) for logging. You should be able to specify [your own configuration file](https://tinylog.org/v2/configuration/) via the command line:
`java -jar -Dtinylog.configuration=/tinylog/configuration.properties application.jar`
Make sure you set `org.diozero.level` to `info` or `warn`.

PauMAVA, 2020-06-18:
Thanks! Problem solved 😄

closed 2020-06-18 by PauMAVA