mattjlewis/diozero #59: Current libws281xj.so library not compatible with 8GB Raspberry Pi 4 & Pi 400

treitmaier, 2020-12-03:
To get diozero-ws281x-java running on my 8GB Raspberry Pi 4 Model B Rev 1.4 (d03114) I had to recompile the `libws281xj.so` native library against the latest version of [rpi_ws281x](https://github.com/jgarff/rpi_ws281x). Before recompiling I was getting a Hardware revision is not supported error.

Could you please recompile & update the `libws281xj.so` library to ensure that diozero-ws281x-java is compatible with the new hardware revisions that have been released (e.g. 8GB Pi4, Raspberry Pi 400) since the `diozero-ws281x-java/src/main/resources/lib/libws281xj.so` was last updated.

mattjlewis, 2020-12-04:
Thanks - just pushing a new 1.0.0 snapshot build right now that includes recompiled diozero ws281x native libraries. Please let me know if this works.

treitmaier, 2021-01-11:
The 1.0.0 release fixed this issue.

closed 2021-01-11 by treitmaier