mattjlewis/diozero #60: Problem switching to version 1.0.0

cbockermann, 2021-01-11:
Hi Matt, hi others,

thanks for this great library! I am using it for quite a long time and setup a Java based RFID kid's radio with it.
Unfortunately, switching to 1.0.0 broke my code. I started digging into it, but failed.

I am using the MFRC522 Reader on a Raspberry PI Model B Rev 2, with the RST line connected to GPIO25.
With version 0.14 everything works fine.

With version 1.0.0, I get the following error:

```
2021-01-09 04:52:29,596 INFO  main net.scinotes.Rfid - Creating RFID reader instance...
Exception in thread "main" java.lang.IllegalArgumentException: Chip not defined for pin PinInfo [keyPrefix=GPIO, header=DEFAULT, deviceNumber=25, physicalPin=22, name=GPIO25, chip=-1, lineOffset=-1, modes=[DIGITAL_INPUT, DIGITAL_OUTPUT]]
	at com.diozero.internal.provider.builtin.DefaultDeviceFactory.createDigitalOutputDevice(DefaultDeviceFactory.java:247)
	at com.diozero.internal.spi.GpioDeviceFactoryInterface.provisionDigitalOutputDevice(GpioDeviceFactoryInterface.java:75)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:101)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:85)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:72)
	at com.diozero.devices.MFRC522.<init>(MFRC522.java:263)
	at net.scinotes.hbox.rfid.RfidReader.<init>(RfidReader.java:37)
	at net.scinotes.Rfid.run(Rfid.java:34)
	at net.scinotes.Main.run(Main.java:89)
	at net.scinotes.Main.main(Main.java:93)
```

I am actually not sure, I everything should continue running as with version 0.14 or if some defaults have changed, which I would need to adjust.

Any idea?

Best regards,
      Christian

mattjlewis, 2021-01-12:
Hi Christian, sorry about this. I suspect I know what is going on here, could you let me know the output of these commands please:
```
uname -a
cat -v /proc/device-tree/compatible
ls -l /dev/gpio*
```
In the meantime could you please try run with `-Ddiozero.gpio.chardev=false` and let me know if that helps.

cbockermann, 2021-01-15:
Hi Matt,

thanks for the reply. The `diozero.gpio.chardev=false` settings solved it.

As for the information you asked for:

```
chris@web-radio:~$ uname -a
Linux web-radio 5.4.79+ #1373 Mon Nov 23 13:18:15 GMT 2020 armv6l GNU/Linux
```

```
chris@web-radio:~$ cat -v /proc/device-tree/compatible
raspberrypi,model-b^@brcm,bcm2835^
```

```
chris@web-radio:~$ grep Revision /proc/cpuinfo 
Revision	: 000f
```

```
chris@web-radio:~$ ls -l /dev/gpio*
crw-rw---- 1 root gpio 254, 0 Jan  9 23:17 /dev/gpiochip0
crw-rw---- 1 root gpio 246, 0 Jan  9 23:17 /dev/gpiomem
```

Yes, it's an older PI, Java startup is sloooow, but RFID-Card reading seems to be much faster with MFRC522 using diozero than the Python libs I've tested... (+ I am a Java guy... :-))

Thanks for the library!

Regards,
     Christian

mattjlewis, 2021-01-16:
I've just released 1.0.1 - hopefully with this release you won't need to run with -Ddiozero.gpio.chardev=false

cbockermann, 2021-01-16:
Thanks! Will start testing right away tomorrow and give you feedback.

Christian

cbockermann, 2021-01-17:
Version 1.0.1 works like charm (**without** -Ddiozero.gpio.chardev=false).

Thanks for the quick fix and new release!

mattjlewis, 2021-02-26:
You're very welcome.

closed 2021-02-26 by mattjlewis