mattjlewis/diozero #61: Problem with version 1.1.1  Error looking up methodID for com/diozero/internal/provider/builtin/gpio/GpioLineEventListener.event(IIJ)V

pezi, 2021-02-12:
I tested the BME280 with version1.1.1, but I run into following error - with 1.1.0 my code works

`
Error looking up methodID for com/diozero/internal/provider/builtin/gpio/GpioLineEventListener.event(IIJ)V
18:20:16.521 [main] WARN com.diozero.util.LibraryLoader.loadLibrary - Error loading library 'diozero-system-utils' from classpath, trying System.loadLibrary: event
Error looking up methodID for com/diozero/internal/provider/builtin/gpio/GpioLineEventListener.event(IIJ)V
18:20:16.548 [main] ERROR com.diozero.util.LibraryLoader.loadLibrary - Error loading library 'diozero-system-utils' from system library path: java.lang.NoSuchMethodError: event
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'com.diozero.internal.provider.builtin.gpio.GpioChip com.diozero.internal.provider.builtin.gpio.NativeGpioDevice.openChip(java.lang.String)'
        at com.diozero.internal.provider.builtin.gpio.NativeGpioDevice.openChip(Native Method)
        at com.diozero.internal.provider.builtin.gpio.GpioChip.lambda$openAllChips$1(GpioChip.java:67)
        at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
        at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:177)
        at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
        at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
        at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:913)
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
        at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:578)
        at com.diozero.internal.provider.builtin.gpio.GpioChip.openAllChips(GpioChip.java:68)
        at com.diozero.internal.provider.builtin.DefaultDeviceFactory.start(DefaultDeviceFactory.java:100)
        at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:85)
        at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:108)
        at com.diozero.api.I2CDevice$Builder.build(I2CDevice.java:145)
        at com.diozero.devices.BME280.<init>(BME280.java:281)
        at com.diozero.devices.BME280.<init>(BME280.java:266
`

mattjlewis, 2021-02-12:
Ah, apologies I know the problem here. Will push 1.1.2 ASAP.

pezi, 2021-02-12:
Thanks - I know how much effort is to create a project documentation - I found some minor link issues on ttps://github.diozero.com/	.

https://github.diozero.com/	
...Raspberry Pi (all versions). (bad link)	
		
https://github.diozero.com/application.html		
...  See my webapp project for an example. 
https://github.com/mattjlewis/diozero/blob/master/diozero-webapp/pom.xml (not found)

mattjlewis, 2021-02-12:
Just released 1.1.2 to fix this issue - invalid native method reference in my C code; very bad testing on my behalf. Hopefully this resolves your issue.

pezi, 2021-02-14:
This fix solves the reason  of this issues.

closed 2021-02-14 by pezi