mattjlewis/diozero #62: BME280.java - humidity value is too high

pezi, 2021-02-15:
This code shows  76 % for humidity in my test run.

3 other Tests

1.) https://raw.githubusercontent.com/ControlEverythingCommunity/BME280/master/Java/BME280.java
shows 40%
2.) python BME280 40%
3.) python SHT31 sensor 38%

And very strange - after calling the python BME280 code once - the BME280 sensor has reached an internal state that the blamed code works correct. I spend some hours to nail down the problem. But I am an average pyhton programmer.

I will try to fix this problem.

mattjlewis, 2021-02-15:
It looks like an issue with the order in which the configuration registers are written. I replicated the issue locally (ran my code and got 63.37% RH, ran the Pi4j one that you linked to and got 55.44% RH; I then reran my version and get 55.40% RH. The only difference that I could see is that I was writing the humidity control register after the control measure register. I swapped these two lines around so that it writes the config registers in the same order, rebooted and it appears to be ok now. I presume this is documented in the data sheet - didn't expect it to be so fussy!
Good spot, I'll commit and push as v1.1.4 shortly.

pezi, 2021-02-17:
Thanks for fixing the problem - I found explicit a hint in the code, C or python, to write the registers in the right order. I tried it, but didn't worked for me - BUT at this time your BME280 code base was strongly modified by me trying to fix the problem.

closed 2021-02-23 by pezi