mattjlewis/diozero #64: BME680.java - wrong implementation

pezi, 2021-03-12:
At first, thanks for your great work! 

But I think, the code 
https://github.com/mattjlewis/diozero/blob/master/diozero-core/src/main/java/com/diozero/devices/BME680.java
has following problems:

``` Java
public void setTemperatureOversample(OversamplingMultiplier value) {
		setRegByte(CONFIG_T_P_MODE_ADDRESS, (byte) OVERSAMPLING_TEMPERATURE_MASK, OVERSAMPLING_TEMPERATURE_POSITION,
				value.ordinal())
```

You defined  
``` Java
public enum OversamplingMultiplier {
		NONE(0), X1(1), X2(2), X4(4), X8(8), X16(16);

		public int getCycles() {
			return cycles;
		}
```

I think you intended to you use value.getCycles() instead of value.ordinal().

``` Java
public void setTemperatureOversample(OversamplingMultiplier value) {
		setRegByte(CONFIG_T_P_MODE_ADDRESS, (byte) OVERSAMPLING_TEMPERATURE_MASK, OVERSAMPLING_TEMPERATURE_POSITION,
				value.getCycles())
```

mattjlewis, 2021-03-12:
The code is correct I believe. If you check p30 of the [data sheet](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme680-ds001.pdf) you will see that the device expects values 0..5 hence the use of ordinal rather than cycles.
Happy to be corrected if I am reading it wrong.

pezi, 2021-03-12:
You are right! I didn't looked at the data sheet - I was confused by the the explizit cycle defintions. Sorry!

closed 2021-03-12 by pezi