mattjlewis/diozero #66: Operation on RPi Compute 4 with Pigpio

neil-benn, 2021-03-19:
Hello,

  I've got an RPi compute module 4 running (CM4) 32 bit RPiOS which I've tested with pigpio in python and it works just fine.  I'm now trying to use diozero with the pigpio provider.  My pom dependencies are:

	<dependencies>
		
		<dependency>
			<groupId>com.diozero</groupId>
			<artifactId>diozero-core</artifactId>
			<version>1.1.6</version>
		</dependency>
		
		<dependency>
			<groupId>com.diozero</groupId>
			<artifactId>diozero-provider-pigpio</artifactId>
			<version>1.1.6</version>
		</dependency>
	</dependencies>

  The code is dead simple; just trying to read from GPIO 17:

package com.ziath.bitbang;

import com.diozero.api.DigitalInputDevice;

public class TestGpio {

	public TestGpio() throws InterruptedException {
	}
	
	public static void main(String args[]) {
		try(DigitalInputDevice gpio = new DigitalInputDevice(17)){
			System.out.println(gpio.getValue());
		}
	}
}

  However when I run this on the CM4 I get the following:

pi@rpi-compute-32-1:~ $ appassembler/bin/testgpio
Error in gpioInitialise: -1
Exception in thread "main" java.util.ServiceConfigurationError: com.diozero.internal.spi.NativeDeviceFactoryInterface: Provider com.diozero.internal.provider.pigpioj.PigpioJDeviceFactory could not be instantiated
        at java.base/java.util.ServiceLoader.fail(ServiceLoader.java:581)
        at java.base/java.util.ServiceLoader$ProviderImpl.newInstance(ServiceLoader.java:803)
        at java.base/java.util.ServiceLoader$ProviderImpl.get(ServiceLoader.java:721)
        at java.base/java.util.ServiceLoader$3.next(ServiceLoader.java:1394)
        at java.base/java.util.Spliterators$IteratorSpliterator.tryAdvance(Spliterators.java:1812)
        at java.base/java.util.stream.ReferencePipeline.forEachWithCancel(ReferencePipeline.java:127)
        at java.base/java.util.stream.AbstractPipeline.copyIntoWithCancel(AbstractPipeline.java:502)
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:488)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
        at java.base/java.util.stream.FindOps$FindOp.evaluateSequential(FindOps.java:150)
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
        at java.base/java.util.stream.ReferencePipeline.findFirst(ReferencePipeline.java:543)
        at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:81)
        at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:114)
        at com.diozero.api.DigitalInputDevice.<init>(DigitalInputDevice.java:116)
        at com.ziath.bitbang.TestGpio.main(TestGpio.java:11)
Caused by: java.lang.RuntimeException: Error initialising pigpio: -1
        at uk.pigpioj.PigpioJNI.<init>(PigpioJNI.java:56)
        at uk.pigpioj.PigpioJ.getImplementation(PigpioJ.java:29)
        at com.diozero.internal.provider.pigpioj.PigpioJDeviceFactory.<init>(PigpioJDeviceFactory.java:67)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
        at java.base/java.util.ServiceLoader$ProviderImpl.newInstance(ServiceLoader.java:779)

  Pigpiod is running:

pi@rpi-compute-32-1:~ $ ps -elf | grep pigpiod
5 S root       498     1  7  80   0 -  3393 -      12:37 ?        00:01:26 pigpiod
0 S pi         977   477  0  80   0 -  1837 pipe_w 12:58 pts/0    00:00:00 grep --color=auto pigpiod

  Is there something further I need to do to get pigpio working with diozero; I need pigpio because I'm going to use this for bitbanging so I need a library that can provide high frequency gpio access.

  Thanks.

Regards,

Neil

mattjlewis, 2021-03-19:
Hi Neil, do you mind retrying with the pigpiod process stopped please.

mattjlewis, 2021-03-19:
BTW, diozero internally has very high performance GPIO control, faster than via pigpioj. This is due to the direct memory mapped GPIO control available with diozero itself; pigpio itself is extremely fast unfortunately the JNI layer adds a bit of an overhead. I have some performance test results that I need to publish - those currently published in the docs are a bit old. From memory, on a Pi4 I got about 2MHz via diozero with pigpio and about 20MHz with diozero itself.

neil-benn, 2021-03-19:
That is very interesting; I tried that but had an issue in that the map didn't work on the RPi4 compute module.  The stack trace for that is below:

pi@rpi-compute-1:~ $ sudo appassembler/bin/testgpio
15:04:52.242 [main] WARN com.diozero.internal.board.raspberrypi.RaspberryPiBoardInfoProvider.lookupByRevision - Unknown Pi model: 20
15:04:52.305 [main] WARN com.diozero.util.LibraryLoader.loadLibrary - Error loading library 'diozero-system-utils' from classpath, trying System.loadLibrary: /tmp/libdiozero-system-utils4780062924207076195.so: /tmp/libdiozero-system-utils4780062924207076195.so: wrong ELF class: ELFCLASS32 (Possible cause: architecture word width mismatch)
15:04:52.308 [main] ERROR com.diozero.util.LibraryLoader.loadLibrary - Error loading library 'diozero-system-utils' from system library path: java.lang.UnsatisfiedLinkError: no diozero-system-utils in java.library.path: [/usr/java/packages/lib, /usr/lib/aarch64-linux-gnu/jni, /lib/aarch64-linux-gnu, /usr/lib/aarch64-linux-gnu, /usr/lib/jni, /lib, /usr/lib]
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'com.diozero.internal.provider.builtin.gpio.GpioChip com.diozero.internal.provider.builtin.gpio.NativeGpioDevice.openChip(java.lang.String)'
        at com.diozero.internal.provider.builtin.gpio.NativeGpioDevice.openChip(Native Method)
        at com.diozero.internal.provider.builtin.gpio.GpioChip.lambda$openAllChips$1(GpioChip.java:66)
        at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
        at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:177)
        at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
        at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
        at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:913)
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
        at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:578)
        at com.diozero.internal.provider.builtin.gpio.GpioChip.openAllChips(GpioChip.java:68)
        at com.diozero.internal.provider.builtin.DefaultDeviceFactory.start(DefaultDeviceFactory.java:100)
        at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:89)
        at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:114)
        at com.diozero.api.DigitalInputDevice.<init>(DigitalInputDevice.java:116)
        at com.ziath.bitbang.TestGpio.main(TestGpio.java:11)
pi@rpi-compute-1:~ $

I can't see where this library would come from?

neil-benn, 2021-03-19:
This is on a 32 bit OS, note that it is a RPi 4 Compute module which could cause the problem for the pins?

pi@rpi-compute-32-1:~ $ sudo appassembler/bin/testgpio
15:56:38.783 [main] WARN com.diozero.internal.board.raspberrypi.RaspberryPiBoardInfoProvider.lookupByRevision - Unknown Pi model: 20
15:56:38.924 [main] WARN com.diozero.internal.provider.builtin.DefaultDeviceFactory.lambda$start$0 - Configured pin name (GLOBAL_RESET) doesn't match that detected (ANT1)
15:56:38.925 [main] WARN com.diozero.internal.provider.builtin.DefaultDeviceFactory.lambda$start$0 - Configured pin name (SD_OC_N) doesn't match that detected (ANT2)
true
15:56:38.963 [main] WARN com.diozero.util.EpollNative.deregister - No file descriptor for '/sys/class/gpio/gpio17/value'

neil-benn, 2021-03-19:
I tried a 'normal' RPi4 on 64 bit OS; same problem.  However, I tried the 'normal' RPi4 on 32 bit OS and we have success:

true
16:23:49.701 [main] WARN com.diozero.util.EpollNative.deregister - No file descriptor for '/sys/class/gpio/gpio17/value'

Minor problem with the WARN and the file referenced isn't there - that is correct.

Is there a way to add in the descriptor for RPi Compute Modeul 4 and can we compile the native libraries against a 64 bit OS?

mattjlewis, 2021-03-19:
I've just ordered a CM4 so that I can test myself and resolve those library issues. diozero contains a small native library (diozero-system-utils) for interfacing with I2C / SPI / GPIO etc. From the stack trace it would appear that it is attempting to load a 32-bit library on a 64 bit O/S:
```
wrong ELF class: ELFCLASS32 (Possible cause: architecture word width mismatch).
no diozero-system-utils in java.library.path: [/usr/java/packages/lib, /usr/lib/aarch64-linux-gnu/jni, /lib/aarch64-linux-gnu, /usr/lib/aarch64-linux-gnu, /usr/lib/jni, /lib, /usr/lib]
```
Did pigpioj work ok when pigpiod was shutdown?

While I wait for my CM4 to arrive, do you mind running the following commands and sharing the output please:
```
tail -4 /proc/cpuinfo
cat -v /proc/device-tree/compatible
cat -v /proc/device-tree/model
uname -a
gpioinfo
```

mattjlewis, 2021-03-19:
Plus diozero _should_ work on 64-bit O/S, I'm just booting my Pi4 Ubuntu and will test locally. Looks like an error with my latest cross compilation.

neil-benn, 2021-03-19:
This is on an RPi Compute Module 4 running 64 bit (let me know if you want 32 bit as well), please see below:

pi@rpi-compute-1:~ $ tail -4 /proc/cpuinfo
Hardware        : BCM2835
Revision        : a03140
Serial          : 10000000b68ef68d
Model           : Raspberry Pi Compute Module 4 Rev 1.0

pi@rpi-compute-1:~ $ cat -v /proc/device-tree/compatible
raspberrypi,4-compute-module^@brcm,bcm2711^@

pi@rpi-compute-1:~ $ cat -v /proc/device-tree/model
Raspberry Pi Compute Module 4 Rev 1.0^@

pi@rpi-compute-1:~ $ uname -a
Linux rpi-compute-1 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux

pi@rpi-compute-1:~ $ gpioinfo
gpiochip0 - 54 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused   input  active-high
        line   2:      unnamed       unused   input  active-high
        line   3:      unnamed       unused   input  active-high
        line   4:      unnamed       unused   input  active-high
        line   5:      unnamed       unused   input  active-high
        line   6:      unnamed       unused   input  active-high
        line   7:      unnamed       unused   input  active-high
        line   8:      unnamed       unused   input  active-high
        line   9:      unnamed       unused   input  active-high
        line  10:      unnamed       unused   input  active-high
        line  11:      unnamed       unused   input  active-high
        line  12:      unnamed       unused   input  active-high
        line  13:      unnamed       unused   input  active-high
        line  14:      unnamed       unused   input  active-high
        line  15:      unnamed       unused   input  active-high
        line  16:      unnamed       unused   input  active-high
        line  17:      unnamed       unused   input  active-high
        line  18:      unnamed       unused   input  active-high
        line  19:      unnamed       unused   input  active-high
        line  20:      unnamed       unused   input  active-high
        line  21:      unnamed       unused   input  active-high
        line  22:      unnamed       unused   input  active-high
        line  23:      unnamed       unused   input  active-high
        line  24:      unnamed       unused   input  active-high
        line  25:      unnamed       unused   input  active-high
        line  26:      unnamed       unused   input  active-high
        line  27:      unnamed       unused   input  active-high
        line  28:      unnamed       unused   input  active-high
        line  29:      unnamed       unused   input  active-high
        line  30:      unnamed       unused   input  active-high
        line  31:      unnamed       unused   input  active-high
        line  32:      unnamed       unused   input  active-high
        line  33:      unnamed       unused   input  active-high
        line  34:      unnamed       unused   input  active-high
        line  35:      unnamed       unused   input  active-high
        line  36:      unnamed       unused   input  active-high
        line  37:      unnamed       unused   input  active-high
        line  38:      unnamed       unused   input  active-high
        line  39:      unnamed       unused   input  active-high
        line  40:      unnamed       unused   input  active-high
        line  41:      unnamed       unused   input  active-high
        line  42:      unnamed       "led0"  output  active-high [used]
        line  43:      unnamed       unused   input  active-high
        line  44:      unnamed       unused   input  active-high
        line  45:      unnamed       unused   input  active-high
        line  46:      unnamed       unused   input  active-high
        line  47:      unnamed       unused   input  active-high
        line  48:      unnamed       unused   input  active-high
        line  49:      unnamed       unused   input  active-high
        line  50:      unnamed       unused   input  active-high
        line  51:      unnamed       unused   input  active-high
        line  52:      unnamed       unused   input  active-high
        line  53:      unnamed       unused   input  active-high
gpiochip1 - 8 lines:
        line   0:      "BT_ON"       unused  output  active-high
        line   1:      "WL_ON"       unused  output  active-high
        line   2: "PWR_LED_OFF" "led1" output active-low [used]
        line   3:       "ANT1"       "ant1"  output  active-high [used]
        line   4: "VDD_SD_IO_SEL" "vdd-sd-io" output active-high [used]
        line   5:   "CAM_GPIO"       unused  output  active-high
        line   6:  "SD_PWR_ON" "sd_vcc_reg"  output  active-high [used]
        line   7:       "ANT2"       "ant2"  output  active-high [used]

Let me know if you want anything else

mattjlewis, 2021-03-19:
1.1.7 just released, please let me know how you get on. I've added detection logic for the Pi400 and CM4 plus rebuilt the native library for aarch64 - I've done some basic tests on my Pi4B using 64bit Ubuntu.
I think you are testing both 32bit and 64bit as well as with and without pigpioj. I'd be very keen to see your results, hopefully my CM4 will arrive soon. Also keen to hear what you are developing on the CM4 but that's a separate conversation.

neil-benn, 2021-03-19:
Hello,

  Thats great thanks, the setup is on my desk at work and I had to get two
small kids to bed so no chance to check it out.  I'll catch up tomorrow,
don't forget on the CM4 you need an IO board.  Don't cross the 3.3V and 5V
pins, you get a nice smoke cloud!

Cheers,

Neil

neil-benn, 2021-04-08:
Sorry to be quiet for a while; I've been waiting for our oscilloscope.  At the moment with just toggling one pin using the mmap the max speed I can get on a 32 bit raspberry pi 4 is 3MHz, the code is as follows:

	public void toggleClock(int count, int channel) {
		fpga_reset.on();
		fpga_reset.off();
		rts.on();
		for (int i=0; i< count; i++) {
			clockPins[channel].toggle();
		}
		rts.off();
	}

Ignore the fpga reset pins but I'm just measuring between the rts pin going high and low - the count is at 240,000 cycles.  I'm far below the goal of 20MHz; am I just being limited by the speed of the Java app running on the processor?

mattjlewis, 2021-04-08:
I need to publish my latest performance results - I'm holding back as there are quite a few combinations to test.
On my Pi4 running RaspiOS 64-bit (I get similar results on the 32-bit RaspiOS):
```
uname -a
Linux pi4-64 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
```
I get ~3.5MHz on a Pi4 using the pigpio provider. I believe the relatively poor performance is down to JNI:
```
> sudo java -cp diozero-sampleapps-1.1.9.jar:diozero-provider-pigpio-1.1.9.jar com.diozero.sampleapps.perf.GpioPerfTest 21 5000000
13:48:34.005 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 5,000,000 iterations: 1.922 s, frequency: 2,601,457 Hz
13:48:35.939 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 5,000,000 iterations: 1.913 s, frequency: 2,613,696 Hz
13:48:37.847 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 5,000,000 iterations: 1.907 s, frequency: 2,621,919 Hz
13:48:39.756 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 5,000,000 iterations: 1.908 s, frequency: 2,620,545 Hz
13:48:41.664 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 5,000,000 iterations: 1.907 s, frequency: 2,621,919 Hz
```
I get ~27.5MHz on a Pi4 using the built-in default provider (which uses mmap under the hood):
```
> java -cp diozero-sampleapps-1.1.9.jar com.diozero.sampleapps.perf.GpioPerfTest 21 50000000
13:47:18.593 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.799 s, frequency: 27,793,218 Hz
13:47:20.436 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.821 s, frequency: 27,457,441 Hz
13:47:22.233 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.795 s, frequency: 27,855,153 Hz
13:47:24.031 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.797 s, frequency: 27,824,151 Hz
13:47:25.832 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.799 s, frequency: 27,793,218 Hz
```

neil-benn, 2021-04-08:
Interestingly you are using the unsafe call to set the pin; obviously the method name puts me off but what does it do in reality, can I use it in production?

mattjlewis, 2021-04-08:
Yes the method is perfectly "safe", possibly a bad name. The on() and off() methods call setValueUnsafe(). From the Javadoc:
```
	/**
	 * Unsafe operation that has no synchronisation checks and doesn't compensate
	 * for active low logic. Included primarily for performance tests.
	 * 
	 * @param value The new value
	 * @throws RuntimeIOException If an I/O error occurs
	 */
```
I will do a quick test with on and off to see the difference.

mattjlewis, 2021-04-08:
27.5MHz when using on / off:
```
> java -cp diozero-sampleapps-1.2.0.jar com.diozero.sampleapps.perf.GpioPerfTest 21 50000000
14:32:39.700 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.840 s, frequency: 27,173,913 Hz
14:32:41.552 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.831 s, frequency: 27,307,482 Hz
14:32:43.351 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.798 s, frequency: 27,808,676 Hz
14:32:45.161 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.808 s, frequency: 27,654,867 Hz
14:32:46.961 [main] INFO com.diozero.sampleapps.perf.GpioPerfTest.test - Duration for 50,000,000 iterations: 1.798 s, frequency: 27,808,676 Hz
```

neil-benn, 2021-04-08:
This is interesting; I copied your perf code and ran that and there seems to almost be a 'warm up' period; I have the results below:

pi@rpi4-1:~ $ sudo appassembler/bin/testgpio
Duration for 240,000 iterations: 0.041 s, frequency: 5,853,658.537 Hz
Duration for 240,000 iterations: 0.037 s, frequency: 6,486,486.486 Hz
Duration for 240,000 iterations: 0.01 s, frequency: 24,000,000 Hz
Duration for 240,000 iterations: 0.01 s, frequency: 24,000,000 Hz
Duration for 240,000 iterations: 0.01 s, frequency: 24,000,000 Hz

Obviously because we are timing in milliseconds the last two elements have simply exceeded the calculation limit; I could try nonTime but TBH 24 mill versus 27 mill isn't worth worrying about!

  Does this 'warm up' register with anything you can think of?

mattjlewis, 2021-04-08:
This should be just standard JVM JIT behaviour. E.g. https://www.baeldung.com/java-jvm-warmup
I have wondered whether it would be possible to use the GraalVM with diozero (https://www.graalvm.org). This wouldn't make things faster (possibly slower overall), just more predictable.

neil-benn, 2021-04-08:
The warmup is indeed affecting us; the timings below show that:

Iteration 0 - Sent 240K ints in 113 millis
Iteration 1 - Sent 240K ints in 80 millis
Iteration 2 - Sent 240K ints in 52 millis
Iteration 3 - Sent 240K ints in 51 millis
Iteration 4 - Sent 240K ints in 51 millis
Iteration 5 - Sent 240K ints in 51 millis

  That gives me 4.7MHz; note that I'm actually doing calculations to extract the bits from an int array and then sending the bits across 8 * 2 GPIO pins in two threads with a control pin for each channel.  Previously doing the above with just one thread was giving me about 50% on top of the above times.  That was on the 32 OS with the RPi4; just putting together the 64 bit ubuntu card.

neil-benn, 2021-04-08:
Interestingly the 64 bit version is slower!

Iteration 0 - Sent 240K ints in 114 millis
Iteration 1 - Sent 240K ints in 87 millis
Iteration 2 - Sent 240K ints in 84 millis
Iteration 3 - Sent 240K ints in 84 millis
Iteration 4 - Sent 240K ints in 85 millis
Iteration 5 - Sent 240K ints in 85 millis
Iteration 6 - Sent 240K ints in 85 millis
Iteration 7 - Sent 240K ints in 84 millis
Iteration 8 - Sent 240K ints in 84 millis
Iteration 9 - Sent 240K ints in 85 millis

I strongly suspect that this is a 64 bit job relating to converting ints to bytes to bits though

neil-benn, 2021-04-08:
Hello,

  Teh 64 bit version is a bit quicker:

Iteration 0 - Sent 240K ints in 97 millis
Iteration 1 - Sent 240K ints in 59 millis
Iteration 2 - Sent 240K ints in 45 millis
Iteration 3 - Sent 240K ints in 45 millis
Iteration 4 - Sent 240K ints in 45 millis
Iteration 5 - Sent 240K ints in 44 millis
Iteration 6 - Sent 240K ints in 54 millis
Iteration 7 - Sent 240K ints in 45 millis
Iteration 8 - Sent 240K ints in 44 millis
Iteration 9 - Sent 240K ints in 45 millis

  I'm clearly constrained by my logic in converting the bytes to the pins rather than the diozero speed.  Moving to the CM4 test platform now.

  Thanks - I normally am either talking to lawyers or handling larger spring boot applications - this is good fun!

Cheers,

Neil

mattjlewis, 2021-04-09:
Fairly different to my day job too.

FYI I did some investigation myself, in particular the use of GraalVM to avoid the "warm-up" issues relating to the JVM JIT compiler. Documented here: https://github.diozero.com/performance/graalvm.html

neil-benn, 2021-04-13:
Hello,

  That Graal JVMCICompiler seems interesting - I'm off for most of the rest
of the week (vaccine on Thursday!) but I'll attempt to recreate it here.
The java process is running a large spring boot app with lots of other
things happening but if the gpio is really that fast we could get some
significant speed boost.

Cheers,

Neil

-- 

Neil Benn MSc
Ziath Ltd
Phone: +44 (0) 1223 855021
http://www.ziath.com

*Please consider the environment before printing this email.*

Follow us on Facebook <http://www.facebook.com/ziathltd>, Twitter
<http://www.twitter.com/ziath_ltd> or LinkedIn
<http://www.linkedin.com/company/ziath-ltd>

IMPORTANT NOTICE: This message, including any attached documents, is
intended only for the use of the individual or entity to which it is
addressed, and may contain information that is privileged, confidential and
exempt from disclosure under applicable law. If the reader of this message
is not the intended recipient, or the employee or agent responsible for
delivering the message to the intended recipient, you are hereby notified
that any dissemination, distribution or copying of this communication is
strictly prohibited. If you have received this communication in error,
please notify Ziath Ltd immediately by email at ***@***.*** Thank you.

mattjlewis, 2021-05-13:
FYI I've produced a new article on Medium.com about GraalVM plus made a few updates to the internal docs page linked to previously.
https://derviceiozero.medium.com/building-portable-binaries-for-single-board-computers-with-diozero-and-graalvm-b659f79d5c71
I'll close this issue on the assumption that all is ok. Please open another if you have any other queries.

closed 2021-05-13 by mattjlewis