mattjlewis/diozero #68: Stack smashing detected

NoelVaes, 2021-03-31:
I use Diozero with provider mmap on RaspberryPi 4 with Raspbios 64 bit (arm64).

Since version 1.1.1 I encounter problems (versions up to 1.1.0 do work):
**Version 1.1.1**: 
```
WARN com.diozero.util.LibraryLoader.loadLibrary - Error loading library 'diozero-system-utils' from classpath, trying System.loadLibrary: /tmp/libdiozero-system-utils8754972271102907794.so: /tmp/libdiozero-system-utils8754972271102907794.so: wrong ELF class: ELFCLASS32 (Possible cause: architecture word width mis
match)

```
**Version 1.1.7**:
`*** stack smashing detected ***: <unknown> terminated`

mattjlewis, 2021-03-31:
This is strange - I test on Ubuntu 64bit on my Pi4 which works. Downloading the Raspbios 64 bit beta now and I'll see what I can do.

NoelVaes, 2021-03-31:
I use this base image:
https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/

mattjlewis, 2021-03-31:
Strange:
> unname -a
Linux pi4-64 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
```
> java -cp diozero-sampleapps-1.1.8.jar com.diozero.sampleapps.ButtonTest 5
09:24:01.762 [diozero-GpioChip-processEvents-890992549] INFO com.diozero.sampleapps.ButtonTest.lambda$test$1 - Released
09:24:01.759 [main] INFO com.diozero.sampleapps.ButtonTest.test - Waiting for 10s - *** Press the button connected to input pin 5 ***
09:24:01.779 [diozero-GpioChip-processEvents-890992549] INFO com.diozero.sampleapps.ButtonTest.lambda$test$2 - Event: DigitalInputEvent [gpio=5, epochTime=1617179041756, nanoTime=1159548947718, value=true, active=false]
09:24:02.376 [diozero-GpioChip-processEvents-890992549] INFO com.diozero.sampleapps.ButtonTest.lambda$test$0 - Pressed
09:24:02.377 [diozero-GpioChip-processEvents-890992549] INFO com.diozero.sampleapps.ButtonTest.lambda$test$2 - Event: DigitalInputEvent [gpio=5, epochTime=1617179042375, nanoTime=1160195919078, value=false, active=true]
09:24:02.564 [diozero-GpioChip-processEvents-890992549] INFO com.diozero.sampleapps.ButtonTest.lambda$test$1 - Released
```

I reran the above with the mmap provider (which is no longer required as it is used internally now) and all fine.

I've also tested I2C using the pigpioj provider (which also has native code) and that worked fine too.

This is with diozero 1.1.8.

NoelVaes, 2021-03-31:
Thanks for your quick reply.

The problem is related to MCP23008.

Here is my code:

```
public class DiozeroTest {  
   public static void main(String[] args) throws InterruptedException {  
      System.out.println("Diozero Test");  
      try (MCP23008 relais = new MCP23008(1, 32, MCP23xxx.INTERRUPT_GPIO_NOT_SET)) {  
         relais.setValue(7, true); // Relais 1  
         Thread.sleep(1000);  
         relais.setValue(7, false); // Relais 1  
      }  
   }  
} 
``` 

With **version 1.1.1** I get this result:

```
Diozero Test
15:10:06.630 [main] WARN com.diozero.util.LibraryLoader.loadLibrary - Error loading library 'diozero-system-utils' from classpath, trying System.loadLibrary: /tmp/libdiozero-system-utils9132754380114466564.so: /tmp/libdiozero-system-utils9132754380114466564.so: wrong ELF class: ELFCLASS32 (Possible cause: architecture word width mismatch)
15:10:06.647 [main] ERROR com.diozero.util.LibraryLoader.loadLibrary - Error loading library 'diozero-system-utils' from system library path: java.lang.UnsatisfiedLinkError: no diozero-system-utils in java.library.path: [/usr/java/packages/lib, /usr/lib/aarch64-linux-gnu/jni, /lib/aarch64-linux-gnu, /usr/lib/aarch64-linux-gnu, /usr/lib/jni, /lib, /usr/lib]
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'com.diozero.internal.provider.builtin.gpio.GpioChip com.diozero.internal.provider.builtin.gpio.NativeGpioDevice.openChip(java.lang.String)'
        at com.diozero.internal.provider.builtin.gpio.NativeGpioDevice.openChip(Native Method)
        at com.diozero.internal.provider.builtin.gpio.GpioChip.lambda$openAllChips$1(GpioChip.java:67)
        at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
        at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:177)
        at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
        at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
        at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:913)
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
        at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:578)
        at com.diozero.internal.provider.builtin.gpio.GpioChip.openAllChips(GpioChip.java:68)
        at com.diozero.internal.provider.builtin.DefaultDeviceFactory.start(DefaultDeviceFactory.java:100)
        at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:85)
        at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:108)
        at com.diozero.internal.spi.AbstractDeviceFactory.<init>(AbstractDeviceFactory.java:58)
        at com.diozero.devices.mcp23xxx.MCP23xxx.<init>(MCP23xxx.java:118)
        at com.diozero.devices.mcp23xxx.MCP23x08.<init>(MCP23x08.java:94)
        at com.diozero.devices.MCP23008.<init>(MCP23008.java:63)
        at com.diozero.devices.MCP23008.<init>(MCP23008.java:59)
        at com.vangenechten.test.diozero.DiozeroTest.main(DiozeroTest.java:53)

```

With **version 1.1.8** I get this result:
```
Diozero Test
*** stack smashing detected ***: <unknown> terminated
Aborted
```
With **version 1.1.0** it works fine (and also with preceding versions).

mattjlewis, 2021-03-31:
Ok, I can reproduce on Ubuntu 64-bit as well as RaspiOS 64-bit but not on 32-bit RaspiOS.
The issue appears to be in the native I2C code. If you run with "-Ddiozero.i2c.javaraf" it uses the Java code instead and this worked for me.
There must be a bug in the JNI I2C code - I'll continue to investigate.

mattjlewis, 2021-03-31:
You can ignore those warnings. I've found the error - it's this innocuous bit of code:
	uint32_t funcs;
	int rc = ioctl(fd, I2C_FUNCS, &funcs);
funcs needs to be uint64_t on 64-bit operating systems. I've made the fix and will release in 1.1.9 tonight.

mattjlewis, 2021-03-31:
Thanks for raising!

NoelVaes, 2021-03-31:
Thanks for your quick help and for this great library! We are very satisfied about it.

mattjlewis, 2021-03-31:
Just released 1.1.9 which passes my tests. Please reopen if you encounter any further issues.

wilbertjoosen, 2021-09-25:
Hi, i have the same issue. I'm using beaglebone black with java 1.8 and debian https://debian.beagleboard.org/images/bone-debian-10.3-iot-armhf-2020-04-06-4gb.img.xz

18:54:19.290 [http-nio-8080-exec-1] WARN com.diozero.util.LibraryLoader.loadLibrary - Error loading library 'diozero-system-utils' from classpath, trying System.loadLibrary: /tmp/libdiozero-system-utils3908987808951326247so: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/libdiozero-system-utils3908987808951326247so)
18:54:19.321 [http-nio-8080-exec-1] ERROR com.diozero.util.LibraryLoader.loadLibrary - Error loading library 'diozero-system-utils' from system library path: java.lang.UnsatisfiedLinkError: no diozero-system-utils in java.library.path
2021-09-25 18:54:19.949 ERROR 9730 --- [nio-8080-exec-1] o.a.c.c.C.[.[.[/].[dispatcherServlet]    : Servlet.service() for servlet [dispatcherServlet] in context with path [] threw exception [Handler dispatch failed; nested exception is java.lang.UnsatisfiedLinkError: com.diozero.internal.provider.builtin.gpio.NativeGpioDevice.openChip(Ljava/lang/String;)Lcom/diozero/internal/provider/builtin/gpio/GpioChip;] with root cause

mattjlewis, 2021-10-02:
Hi @wilbertjoosen, this is a different issue and related to glibc version differences. Can you please raise as a separate issue.

closed 2021-10-02 by mattjlewis