mattjlewis/diozero #69: GPIO not work anymore on ThinkerBoard

zixzix, 2021-04-14:
Initialization of any type of device on any gpio no longer works on ThinkerBoard in the latest versions of diozero. It worked perfectly up to version 0.13.
In all versions of diozero, including 1.20, using the following code the board is recognized correctly. 
`DeviceFactoryHelper.getNativeDeviceFactory().getBoardInfo();`

Now it goes into error and I get the following message on all gpio's:

`
java.lang.IllegalArgumentException: Can't find chip for id 5
        at com.diozero.internal.provider.builtin.DefaultDeviceFactory.createDigitalOutputDevice(DefaultDeviceFactory.java:256)
        at com.diozero.internal.spi.GpioDeviceFactoryInterface.provisionDigitalOutputDevice(GpioDeviceFactoryInterface.java:84)
        at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:157)
        at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:141)
        at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:128)
....
`
Maybe there is some problem with the gpio mapping? 
Thank you

mattjlewis, 2021-04-14:
Can you please run `cat /etc/os-release`. I am running Armbian on my TinkerBoard and get this output:
```
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Armbian 21.02.3 Focal"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

gpiodetect output:
```
> sudo gpiodetect
gpiochip0 [gpio0] (24 lines)
gpiochip1 [gpio1] (32 lines)
gpiochip2 [gpio2] (32 lines)
gpiochip3 [gpio3] (32 lines)
gpiochip4 [gpio4] (32 lines)
gpiochip5 [gpio5] (32 lines)
gpiochip6 [gpio6] (32 lines)
gpiochip7 [gpio7] (32 lines)
gpiochip8 [gpio8] (16 lines)
```

diozero GpioReadAll output:
```
sudo java -cp diozero-sampleapps-1.2.0.jar com.diozero.sampleapps.GpioReadAll
Header: DEFAULT
+-----+----------+------+---+--------+----------+--------+---+------+----------+-----+
+ GP# +     Name + Mode + V +  gpiod + Physical + gpiod  + V + Mode + Name     + GP# +
+-----+----------+------+---+--------+----------+--------+---+------+----------+-----+
|     |      3v3 |      |   |        |  1 || 2  |        |   |      | 5v       |     |
| 252 | GPIO8_A4 | Unkn | 0 |  8:4   |  3 || 4  |        |   |      | 5v       |     |
| 253 | GPIO8_A5 | Unkn | 0 |  8:5   |  5 || 6  |        |   |      | GND      |     |
|  17 | GPIO0_C1 |   In | 0 |  0:17  |  7 || 8  |  5:9   | 0 | Unkn | GPIO5_B1 | 161 |
|     |      GND |      |   |        |  9 || 10 |  5:8   | 0 | Unkn | GPIO5_B0 | 160 |
| 164 | GPIO5_B4 |   In | 0 |  5:12  | 11 || 12 |  6:0   | 0 | Unkn | GPIO6_A0 | 184 |
| 166 | GPIO5_B6 | Unkn | 0 |  5:14  | 13 || 14 |        |   |      | GND      |     |
| 167 | GPIO5_B7 | Unkn | 0 |  5:15  | 15 || 16 |  5:10  | 0 | In   | GPIO5_B2 | 162 |
|     |      3v3 |      |   |        | 17 || 18 |  5:11  | 0 | In   | GPIO5_B3 | 163 |
| 257 | GPIO8_B1 | Unkn | 0 |  8:9   | 19 || 20 |        |   |      | GND      |     |
| 256 | GPIO8_B0 | Unkn | 0 |  8:8   | 21 || 22 |  5:19  | 0 | In   | GPIO5_C3 | 171 |
| 254 | GPIO8_A6 | Unkn | 0 |  8:6   | 23 || 24 |  8:7   | 0 | Unkn | GPIO8_A7 | 255 |
|     |      GND |      |   |        | 25 || 26 |  8:3   | 0 | Unkn | GPIO8_A3 | 251 |
| 233 | GPIO7_C1 |   In | 0 |  7:17  | 27 || 28 |  7:18  | 0 | Unkn | GPIO7_C2 | 234 |
| 165 | GPIO5_B5 |   In | 0 |  5:13  | 29 || 30 |        |   |      | GND      |     |
| 168 | GPIO5_C0 |   In | 0 |  5:16  | 31 || 32 |  7:23  | 0 | Unkn | GPIO7_C7 | 239 |
| 238 | GPIO7_C6 | Unkn | 0 |  7:22  | 33 || 34 |        |   |      | GND      |     |
| 185 | GPIO6_A1 | Unkn | 0 |  6:1   | 35 || 36 |  7:7   | 0 | Unkn | GPIO7_A7 | 223 |
| 224 | GPIO7_B0 | Unkn | 0 |  7:8   | 37 || 38 |  6:3   | 0 | Unkn | GPIO6_A3 | 187 |
|     |      GND |      |   |        | 39 || 40 |  6:4   | 0 | Unkn | GPIO6_A4 | 188 |
+-----+----------+------+---+--------+----------+--------+---+------+----------+-----+
```

zixzix, 2021-04-14:
I use TinkerOS

```
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

with "sudo gpiodetect" i get a command not found so I have tried with 
sudo java -cp diozero-sampleapps-1.2.0.jar com.diozero.sampleapps.GpioDetect

```
Exception in thread "main" java.lang.NullPointerException: Cannot invoke "java.util.List.forEach(java.util.function.Consumer)" because the return value of "com.diozero.internal.provider.builtin.gpio.GpioChip.getChips()" is null
        at com.diozero.sampleapps.GpioDetect.main(GpioDetect.java:42)
```

sudo java -cp diozero-sampleapps-1.2.0.jar com.diozero.sampleapps.GpioReadAll

[image]

mattjlewis, 2021-04-14:
Try run with `-Ddiozero.gpio.chardev=false`. Hopefully that will get rid of those errors. Have a think about trying Armbian as it uses a much newer kernel and is actively supported.

zixzix, 2021-04-14:
Thank you!! Seems to work fine, I will do more tests asap 👍

closed 2021-05-13 by mattjlewis