mattjlewis/diozero #70: Issue init GPIO on TinkerBoard

zixzix, 2021-05-06:
Thank you!! Seems to work fine, I will do more tests asap 👍

_Originally posted by @zixzix in https://github.com/mattjlewis/diozero/issues/69#issuecomment-819760978_

mattjlewis, 2021-05-13:
Do you mind trying with 1.2.1 - it included some fixes in this area

zixzix, 2021-05-13:
Hello @mattjlewis ,
as specified in the "test environment section" all tests were performed with version 1.2.1 of diozero

mattjlewis, 2021-05-13:
Sorry - missed that! Will take a look this weekend.

mattjlewis, 2021-05-14:
I think I've reproduced scenarios 1-3 with this test app: https://github.com/mattjlewis/diozero/blob/main/diozero-sampleapps/src/main/java/com/diozero/sampleapps/ShutdownRestartTest.java
Scenario 4 might be TinkerBoard specific, will check later.
I'll commit and push as 1.2.2 this weekend after further tests.

mattjlewis, 2021-05-16:
Could you retry with 1.2.2 please.

mattjlewis, 2021-06-03:
Assume fixed.

closed 2021-06-03 by mattjlewis