mattjlewis/diozero #71: MFRC522, setAntennaGain(..) has no effect

lglowania, 2021-05-24:
I'm trying to increase the reading-range of my MFRC522. By using `MFRC522.getPiccType(sak);` i determined my picc is a MIFARE_1K. The data sheet (https://www.mouser.com/datasheet/2/302/MF1S503x-89574.pdf) says:

*Operating distance up to 100 mm
depending on antenna geometry and
reader configuration*

So i assume the antenna gain setting is available in general. But using `mfrc522.setAntennaGain(..)` seems to do nothing, the reading-range remains at ~20mm, no matter what AntennaGain-enum is passed. And `mfrc522.getAntennaGain()` returns always *null*.

Do you have a hint? Thank you.

mattjlewis, 2021-06-03:
Please test with 1.2.3 and let me know if it fixes the issue.

lglowania, 2021-06-04:
Yes, getAntennaGain() now returns the expected result.

mattjlewis, 2021-06-04:
Excellent, thank you. Did it make a difference in range?

closed 2021-06-10 by mattjlewis

lglowania, 2021-06-11:
Sorry for the late response. I had no time to investigate further. In my experiments it did not make any difference in range, but when  i used `mfrc522.setAntennaGain(AntennaGain.DB_48)` i frequently received read errors when placing the transponder. I don't remember exactly the error-output. It was something like `read error on 0x01`. I will comment my insights here in the future.

commit aa806c7f (2021-05-26): Preparing 1.2.3 - Orange Pi Zero+ / One+, bug fix #71 MFRC522 antenna
gain, use diozero rather than Device I/O Zero - hence update to licence
headers.