mattjlewis/diozero #75: WS281xNative is not 64-bit compatible

treitmaier, 2021-07-05:
When trying to initialise the latest (1.3.0) version of `diozero-ws281x-java` on 64-bit Raspberry Pi 4, I encounter the following error. The same program runs without issues on 32-bit Raspberry Pi 4. 

```
Error loading library 'ws281xj'
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'java.nio.ByteBuffer com.diozero.ws281xj.rpiws281x.WS281xNative.initialise(int, int, int, int, int, int, int)'
	at com.diozero.ws281xj.rpiws281x.WS281xNative.initialise(Native Method)
	at com.diozero.ws281xj.rpiws281x.WS281x.<init>(WS281x.java:168)
	at com.diozero.ws281xj.rpiws281x.WS281x.<init>(WS281x.java:150)
```

mattjlewis, 2021-07-06:
Thanks for raising - I have a fix prepared and will include in the next release. It would be great if you could test by downloading [libws281xj-aarch64.so](https://github.com/mattjlewis/diozero/raw/main/diozero-ws281x-java/src/main/resources/lib/libws281xj-aarch64.so) to your 64-bit Pi, renaming to libws281xj.so and including the containing folder on the Java library path.

mattjlewis, 2021-11-08:
Should now be fixed as of 1.3.1

closed 2021-11-08 by mattjlewis