mattjlewis/diozero #82: switch to slf4j

stbischof, 2021-09-30:
Since April 24, 2021 the SLF4J binding of tinylog supports also SLF4J 2.0 .

would it be an option to switch to the more used SLF4J api and let the user decide about the impl. to everyone can use tinylog ore someone else.

mattjlewis, 2021-10-02:
Thanks for the suggestion. I won't rule this out completely for a future major release (2.0 or 3.0) as I can see the benefits, however, I am reticent to add in the short-term given the huge amount of work for limited value.

EAGrahamJr, 2023-08-19:
In my opinion, the use of `tinylog` makes more sense for "constrained" devices -- thus makes sense as a default. If one has SLF4J dependencies, then just make sure `logback` is **not** included and use `org.tinylog:slf4j-tinylog:$VERSION`. The byte-sizes of the two logging frameworks are staggeringly different.

And, to be honest, `tinylog` has a couple of API bits that **I** think are more functionally useful than `SLF4J` (yeah, I know: heretical).