mattjlewis/diozero #83: BBB Uart Example usage

wilbertjoosen, 2021-11-08:
Hi,

I'm trying to use BBB UART ports, but I haven't been able to yet, I tried to look at the documentation but there is still no data for serial ports. I'm trying to communicate with the SIM 800 module that has the TX and RX. At BBB, I connected UART1 ports on pins 24 and 26

wilbertjoosen, 2021-11-08:
I've just enable uart ports following this: https://stackoverflow.com/questions/35881761/enabling-uart-on-beaglebone-black

mattjlewis, 2021-11-08:
Working ok?

wilbertjoosen, 2021-11-08:
> Working ok?

No, i cant get it working, i need hel for this implementation

mattjlewis, 2021-11-08:
Are you struggling to get UART working on the BBB or have you validated that UART is working but you are struggling with the diozero serial device support?

wilbertjoosen, 2021-11-08:
> Are you struggling to get UART working on the BBB or have you validated that UART is working but you are struggling with the diozero serial device support?

I'm struggling to get UART working, i need suport such as an example.

wilbertjoosen, 2021-11-09:
@mattjlewis 
I'm trying to get it working with this code bellow, is that correct?

`        
SerialDevice serialDevice = SerialDevice.builder("/dev/tty01")
                .setBaud(SerialDevice.BAUD_1152000)
                .setDeviceFilename("/dev/ttyO1")
                .build();

        serialDevice.write("AT".getBytes(StandardCharsets.UTF_8));
`

mattjlewis, 2021-11-11:
The call to setDeviceFilename isn't required and contains an "O" (Oh) instead of a "0" (zero).

Try:
```
SerialDevice serialDevice = SerialDevice.builder("/dev/tty01")
.setBaud(SerialDevice.BAUD_115200)
.build();
serialDevice.write("AT".getBytes(StandardCharsets.UTF_8));
```
Are you able to serial device communication via "/dev/tty01" using other means, e.g. Python?

wilbertjoosen, 2021-11-11:
@mattjlewis thanks for you anwrer. Yerterday i successfuly write an information to my SIM800L with the SerialDevice component, but i have to use Thread.sleep(1000); to wait device to respond like bellow:

        SerialDevice serialDevice = SerialDevice.builder("/dev/ttyO1")
                .setBaud(SerialDevice.BAUD_9600)
                .build();

        serialDevice.write(convert("AT+CMGF=1"));
        Thread.sleep(1000);
        if(serialDevice.bytesAvailable()>0){
            System.out.println("Valor: "+ String.valueOf(serialDevice.readByte()));
        }
        serialDevice.write(convert("AT+CREG?"));
        Thread.sleep(2000);
        serialDevice.write(convert("AT+CMGF?"));
        Thread.sleep(2000);
        serialDevice.write(convert("AT+CMGS=\"XXXXXXXXXXXX\""));
        Thread.sleep(1000);
        serialDevice.write(convert("Message test"));
        Thread.sleep(1000);
        byte[] buffer = new byte[1];
        buffer[0] = 26;
        serialDevice.write(buffer); // send ^z command do end communication after type a message
        Thread.sleep(1000);
        serialDevice.close();

My problem now is to understand how to read from the serial device, i test with the blocK:

 if(serialDevice.bytesAvailable()>0){
            System.out.println("Valor: "+ String.valueOf(serialDevice.readByte()));
 }

 Is thar correct? I'm answering because after any command sent to device an answer is sent and i have to validate it.
In the prints above on the left side i'm watching the interface /dev/ttyO1 and right side are commands i've sent to interface. Now i cant read messages sent by the module, can you show how to do it?
[image]

mattjlewis, 2021-11-11:
I use the underlying NativeSerialDevice class in my [Firmata provider implementation](https://github.com/mattjlewis/diozero/blob/main/diozero-provider-firmata/src/main/java/com/diozero/internal/provider/firmata/adapter/SerialFirmataTransport.java). The [FirmataAdapter](https://github.com/mattjlewis/diozero/blob/main/diozero-provider-firmata/src/main/java/com/diozero/internal/provider/firmata/adapter/FirmataAdapter.java) is responsible for doing the I/O (I have the transport abstraction as communication can also be over the network).
The SerialDevice class that you are using delegates all I/O operations to this class - you should continue to use that one.
One thing to note with my serial device implementation - all reads are blocking operations. I would recommend decouple writes from reads for this reason - I tend to use a BlockingQueue to bridge read and write operations as implemented in the FirmataAdapter.
Hope this helps.

wilbertjoosen, 2021-11-11:
@mattjlewis thaks for your support, it helps a lot

mattjlewis, 2021-11-11:
You're very welcome, let me know if you have any suggestions on how to improve diozero.

closed 2021-11-11 by mattjlewis