mattjlewis/diozero #84: libiio integration

andy-shev, 2021-11-10:
If I missed this, sorry.
Are you planning to have an interface to `libiio`?

mattjlewis, 2021-11-11:
How would you envisage this integration? Using libiio as [diozero "provider"](https://www.diozero.com/concepts.html) that provides access to GPIO / I2C / SPI interfaces or as an alternative mechanism for accessing specific devices completely bypassing the diozero provider layer? From my brief investigation into libiio I suspect the latter is more likely.
I do find the idea very interesting btw. I have been looking into a consistent mechanism for remotely interfacing with devices in a portable manner, in particular embedded devices like the ESP8266 / ESP32 / Pi Pico. The [Firmata protocol](https://github.com/mattjlewis/diozero/tree/main/diozero-provider-firmata) is good but not very efficient.

andy-shev, 2021-11-11:
> How would you envisage this integration? Using libiio as [diozero "provider"](https://www.diozero.com/concepts.html) that provides access to GPIO / I2C / SPI interfaces or as an alternative [...]

Yes, it's a high level interface that allows you to use in-kernel drivers for more than 500 devices (as of today) and counting.

> I have been looking into a consistent mechanism for remotely interfacing with devices in a portable manner, in particular embedded devices like the ESP8266 / ESP32 / Pi Pico. The [Firmata [...]

Yes, I understand, what I'm talking is applicable to Linux running cases.

closed 2022-03-11 by mattjlewis