mattjlewis/diozero #85: Automatic recommendation to use kernel drivers

andy-shev, 2021-11-10:
This (I dunno why I can't write there) https://groups.google.com/g/diozero/c/t3qvWdugoiw makes me wonder, why not to hint people that some of the chips are already supported by the Linux kernel natively like MCP23017 (https://elixir.bootlin.com/linux/latest/source/drivers/pinctrl/pinctrl-mcp23s08_i2c.c).

mattjlewis, 2021-11-11:
Interesting - I was aware that was possible but never tried it. Questions / assumptions:
- are such drivers generally available in the majority of SBC kernels (Raspberry OS, Armbian, ...)?
- if not, do you need to build a custom kernel or can you just compile and install the kernel driver itself?
- if building the kernel driver, I assume it needs to be rebuilt every time there is an update the the SBC's kernel
- also if building a kernel driver, I assume you also need to configure the device tree in order to use it
- are you able to drive GPIOs via the relatively new GPIO character device interface (I have moved away from sysfs for GPIO control)

Apologies if these are basic questions, I tend to just be a user of the Linux kernel rather than building / configuring it.

andy-shev, 2021-11-11:
> Interesting - I was aware that was possible but never tried it. Questions / assumptions:
> 
> * are such drivers generally available in the majority of SBC kernels (Raspberry OS, Armbian, ...)?

Yes. There two problems of different severity:
 - kernel version (should be not too old)
 - actual driver enabled (this is biggest issue since those distros rather _randomly_ choose set of modules they build)

For the first you may talk to Linux kernel device data base: https://cateee.net/lkddb/ (check web UI to see that's up and running).

> * if not, do you need to build a custom kernel or can you just compile and install the kernel driver itself?

Starting from kernel 2.6.x (no need to tell, we are at v5.15 era nowadays) the idea is to make Linux kernel modular, so for most of the modules (esp. when they are simple sensors like bme280) one may just build the driver.

> * if building the kernel driver, I assume it needs to be rebuilt every time there is an update the the SBC's kernel

Right. That's why it's important to tell distro managers that people need drivers.

> * also if building a kernel driver, I assume you also need to configure the device tree in order to use it

Right. Or ACPI.

> * are you able to drive GPIOs via the relatively new GPIO character device interface (I have moved away from sysfs for GPIO control)

As long as you have opened the device, yes. That's true way how it should be done. Currently in the upstream the new libgpiod is under review which supports v2 of ABI.  There is also slowly moving discussion about a D-Bus daemon that allows to have context be kept somewhere else than in application.

> Apologies if these are basic questions, I tend to just be a user of the Linux kernel rather than building / configuring it.

No, it's fine. You are welcome!

mattjlewis, 2021-11-11:
While I have your attention I'd like to clarify terminology. When adding GPIO chardev support I initially used libgpiod but found it a bit restrictive at the time and the [underlying API easier to use](https://github.com/mattjlewis/diozero/blob/main/system-utils-native/src/main/c/com_diozero_internal_provider_builtin_gpio_NativeGpioDevice.c#L104). Is this the GPIO chardev "ABI" that I'm using?

Otherwise, I completely get your point and I need to go do some tests to validate that diozero is able to dynamically discover and connect to GPIO expanders via GPIO chardev. Other, non-GPIO classes of device (e.g. ADCs / DACs) would be quite different - I guess these would expose their own specific interface e.g. via sysfs.

andy-shev, 2021-11-12:
> When adding GPIO chardev support I initially used libgpiod but found it a bit restrictive at the time and the [underlying API easier to [...]

Yes. Ideally somebody should come to `libgpiod` or to create a similart to `libgpiod` (C, Python) and `gpiod` (Golang) APIs for the Java language.

Have you seen, btw, this: https://github.com/sgjava/java-periphery?

> Otherwise, I completely get your point and I need to go do some tests to validate that diozero is able to dynamically discover and connect to GPIO expanders via GPIO chardev. Other, non-GPIO classes [...]

Rather via `libiio`. But you got the idea. It's pity that 100500 projects are repeating the same for the drivers that are exists in the Linux kernel (and some of them for ages).

andy-shev, 2021-11-12:
Ah, what's funny is that one may instantiate I²C peripheral drivers (if they are compiled and exists) via user space. So if we are lucky and kernel in question has a driver, but user missed Device Tree or ACPI description for that, you may enable it (yeah, it will require administrator permissions): https://www.kernel.org/doc/html/latest/i2c/instantiating-devices.html#method-4-instantiate-from-user-space.

sgjava, 2021-11-15:
@andy-shev I've just [committed](https://github.com/sgjava/java-periphery/commit/1eb8f58daff2ae25f2742b278e7fc1b8e37e4058) I2C optimizations so I don't have to deal with i2c_msg struct in Java any more. Also working with Armbian folks to look at integrating c-periphery in armbian-config to add IO capabilities. Userspace is the present and future for device interfaces. I'm using a MCP23017 to replace my old alarm system board using built in magnetic switches.

I've already added c-periphery to U8g2 [arm-linux](https://github.com/olikraus/u8g2/tree/master/sys/arm-linux) port and software I2C is faster then 100 KHz hardware I2C.

mattjlewis, 2021-11-15:
Steve, not sure how that is relevant here (diozero includes a full I2C interface, including smbus and raw I2C transactions). The discussion is whether a library should focus on providing I2C, SPI, GPIO interfaces, vs using kernel drivers such as the MCP23xx7 family of GPIO expansion boards, coupled with relying on a higher-level interface such as libiio - the "100500 projects are repeating the same", i.e. not even going via I2C / SPI / GPIO.

I will investigate this but at the moment I am really concerned that this may actually add more complexity:

- ensuring that appropriate kernel drivers are available, installed and configured. There is a risk that this may also add an additional complexity to the library user to install (maybe even compile) and configure such kernel drivers
- ensuring that libiio supports the particular device
- understanding and implement the libiio interface in Java - the documentation isn't that great from my brief investigations

Implementing SPI / I2C device support for these devices often isn't that complicated and may end up in a simpler solution when you consider all of the above complexities.

sgjava, 2021-11-15:
Andy mentioned it above as a reference for userspace IO, but whatever.

closed 2022-03-11 by mattjlewis