mattjlewis/diozero #90: Alt Fucntion on GPIO

neil-benn, 2022-03-10:
Hello,

  Is there a way to set the alt function of a gpio from within diozero?  I'm trying to keep within diozero as much as possible without resorting to calling system functions.

Cheers,

Neil

mattjlewis, 2022-03-10:
Hi Neil, this would be a great feature enhancement. In the short term I could very easily extend the low-level API [here](https://github.com/mattjlewis/diozero/blob/main/diozero-core/src/main/java/com/diozero/internal/board/raspberrypi/RaspberryPiMmapGpio.java#L231) to allow this. At the moment it only supports setting to digital input / output and PWM. It would be trivial to extend this class to add a method that allows any mode to be set on the assumption that the caller is responsible for specifying a valid mode.

Something like:
```public void setMode(int gpio, int mode);```

To invoke, you would need to get the `MmapGpioInterface` from the `BoardInfo` instance via the `createMmapGpio` operation. There is an example here: https://github.com/mattjlewis/diozero/blob/main/diozero-sampleapps/src/main/java/com/diozero/sampleapps/perf/MmapGpioPerfTest.java#L60

neil-benn, 2022-03-10:
Hello,

  Thanks; I'll take a look at it - struggling with trying to get a dd equivalent working on win 1- at the moment.  It used to work but now brings down PowerShell - grr!!

Cheers,

Neil

closed 2022-08-17 by mattjlewis

neil-benn, 2022-08-17:
Great thanks - I'll wrap this into our app code and update to test.  It'll be a few weeks as I'm off for a couple of weeks.

Feature added to 1.3.4