mattjlewis/diozero #94: ODROID-N2+ support - where should I start?

btajuddin, 2022-05-23:
I have a project that is currently using ODROID-N2+ boards. They're similar to the C2, but not enough to reuse that profile. What pieces do I need in order to build out (and ultimately contribute) a board profile for the N2+? 

I'm seeing several points where I may need to make changes in the code, but I'm afraid I'm going to miss something important. Any sort of list of what I should be implementing would be appreciated. Thanks!

mattjlewis, 2022-05-23:
Do you need high-speed memory mapped GPIO access? If not then it _should_ simply be a matter of understanding the GPIO layout.
Could you please share the contents of the following files please:

- /etc/os-release
- /proc/cpuinfo
- /proc/meminfo
- /proc/device-tree/compatible (please do `cat -v`)
- /proc/device-tree/model (please do `cat -v`)

You can see how these files are used to automatically identify the local board ([LocalSystemInfo class](https://github.com/mattjlewis/diozero/blob/main/diozero-core/src/main/java/com/diozero/sbc/LocalSystemInfo.java)).

The above information will inform the name of the [board definition file](https://github.com/mattjlewis/diozero/tree/main/diozero-core/src/main/resources/boarddefs) that will need to be created for this device. I do need to improve the GPIO layout auto-discovery, unfortunately most kernels do not name the device tree GPIO resources in such a way that they can be automatically detected.

The format of the boarddef file isn't currently documented but is quite straighforward and you should be able to create one yourself from the examples and using the gpioinfo commands. I can guide you through this - the good thing is that it requires no code changes and the file just needs to be on the classpath (in a boarddefs folder).

I assume the board has a [S922X CPU](https://dn.odroid.com/S922X/ODROID-N2/Datasheet/S922X_Public_Datasheet_V0.2.pdf)? I can use that data sheet to help implement memory mapped GPIO control but it is a bit fiddly.

btajuddin, 2022-05-23:
Yeah, I don't need high speed control memory mapped GPIO.

I've created a gist with the information you wanted here: https://gist.github.com/btajuddin/7cf1041a3c1418f0f5d6b55b04374bc5

I've got a full work day ahead of me, but I'll probably take another look at this tonight to try writing the board definition. Should I also be augmenting the OdroidBoardInfoProvider?

btajuddin, 2022-05-23:
I dropped a PR with the basics. I think I am going to need to go to the memory mapped access because I need to manage pull up/pull down settings. I'll take a look at that more later (maybe a separate PR).

They do have some information on the wiki [here](https://wiki.odroid.com/odroid-n2/software/gpio_register_map), so I might be able to start looking at that and testing it on my devices.

btajuddin, 2022-05-24:
Ok, I've done some digging on the mapped memory solution. When I tried to use your native memory map, I was getting an "Invalid argument" error. I'm using `0xFF634400` as the base address (as documented in the processor doc you linked and the wiki I linked) with a block size of 4096.

I tried using Java's memory mapping capability, but I still got a cryptic "Invalid argument" message from the native file truncation routine. I would appreciate any ideas as to why that might be happening. My native programming is pretty rusty.

Here's the exception from my attempt to memory map in Java:

`java.lang.RuntimeException: java.io.IOException: Invalid argument
	at com.diozero.internal.board.odroid.OdroidN2PlusMmapGpio.initialise(OdroidN2PlusMmapGpio.java:60)
	at com.diozero.internal.provider.builtin.DefaultDeviceFactory.start(DefaultDeviceFactory.java:191)
	at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:89)
	at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:112)
	at com.diozero.sampleapps.GpioReadAll.main(GpioReadAll.java:65)
Caused by: java.io.IOException: Invalid argument
	at java.base/sun.nio.ch.FileDispatcherImpl.truncate0(Native Method)
	at java.base/sun.nio.ch.FileDispatcherImpl.truncate(FileDispatcherImpl.java:86)
	at java.base/sun.nio.ch.FileChannelImpl.mapInternal(FileChannelImpl.java:1098)
	at java.base/sun.nio.ch.FileChannelImpl.map(FileChannelImpl.java:1032)
	at com.diozero.internal.board.odroid.OdroidN2PlusMmapGpio.initialise(OdroidN2PlusMmapGpio.java:58)
	... 4 more
`

btajuddin, 2022-05-24:
I figured it out. The offset address isn't aligned to a page, so I tweaked that and it works. I just need to sort out one more error, and I should be able to update the PR with a mmap for the N2.

mattjlewis, 2022-05-24:
This is awesome, nice one. I was just about to check page alignments and you beat me to it. Taking a look at the PR now, looking good so far.

btajuddin, 2022-05-25:
Thanks for merging. I've got things copied over to my project right now, but I'd like to just consume it from maven whenever you get a chance to publish the build. No rush, though.

closed 2022-05-25 by btajuddin